espressif/arduino-esp32 #1864: Ticker active() method is not defined

warwound, 2018-09-15:
This is a simple issue/question.

Why has the Ticker library's active() method not been defined?
I'm using the Ticker library in an ESP8266 project and tried to compile it for ESP32.
Compilation fails with an error:

**undefined reference to `Ticker::active()'**

Looking at the Ticker.h header file i see the active() method but in the Ticker.cpp i see it is not defined:
[https://github.com/esp8266/Arduino/tree/master/libraries/Ticker](url)

Is there a reason for this or is it just an oversight?

warwound, 2018-09-15:
I have just copy/pasted the ESP8266 definition of Ticker::active() into my ESP32 Ticker.cpp and it works perfectly:

`bool Ticker::active(){
	return (bool)_timer;
}`

bertmelis, 2019-01-10:
Oups! :flushed: I just found out I forgot to implement that one.

I was wondering: what is the functionality of `active()`? The solution above works fine for periodic Tickers. But how about the single-shot Tickers?

I would say `active()` means "still running but noy yet fired".

It is relatively simple to implement the functionality, but there are also the PRs for "functional Ticker" and "FreeRTOS style Ticker" so I don't know what is the direction we're heading at.

@me-no-dev  Maybe you can enlighten us?

me-no-dev, 2019-01-10:
we should go the RTOS way. This is what makes sense on ESP32 :)

hreintke, 2019-01-12:
@me-no-dev : 
Can you take a look on #1766.
If that is OK for you I can make a PR to replace current Ticker with this.

bertmelis, 2019-01-12:
:+1: Current ticker uses the legacy timer. And there is no need to maintain multiple implementations imho.

laercionit, 2019-05-20:
I'm porting a code that had the following statement in ESP8266:

Tickert ArrayTicker [10];

Unfortunately the esp32 core is not accepting the ArrayTicker [x] .active () method;
Should this work?

beegee-tokyo, 2019-05-20:
Try
**`Ticker ArrayTicker [10];`**
**instead of
`Tickert ArrayTicker [10];`**

Hope you see the difference.

laercionit, 2019-05-20:
Obviously it was a typo, as mentioned by our friend @warwound, it was missing from the CPP in the method. When copying from ESP8266 worked. Thanks for noticing a typo.

closed 2019-08-14 by stale[bot]

maxgerhardt, 2021-04-26:
This is still not resolved and people in the PlatformIO forum are asking us why their build fails ([here](https://community.platformio.org/t/undefined-reference-to-ticker-active-on-esp32/21005)).

gamebeast2k, 2022-01-05:
yep 9 mon later still not fixed. #5919 resolved the issue.