espressif/arduino-esp32 #1869: "install interrupt handler Failed=261" when initializing two I2C buses

AdrienFelon, 2018-09-18:
### Hardware:
Board:							?Custom board?
Core Installation/update date:			?Release 1.0.0?
IDE name:							?Arduino IDE?
Flash Frequency:					?80Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10?

### Description:
Hello,

I have a custom board with two I2C buses and have been happily using Stickbreaker’s fork (ie. release v.0.2.0 of https://github.com/stickbreaker/arduino-esp32 ) for quite a while. Things have been working great on this fork. Now I’d like to start using Release 1.0.0, since Stickbreaker’s work seems to have been integrated back into the main repo. However, I am hitting lots of error “install interrupt handler Failed=261” when initializing the *second* I2C bus, followed by a crash when setting an hw timer in my application code (cf traces below).

Would anybody have any idea of what I could be doing wrong? Is there something I need to change in my application code for it to work on Release 1.0.0?

From what I can tell, code 261 (0x105 ESP_ERR_NOT_FOUND I think) means “No free interrupt found with the specified flags”. Looking into the code where this error originates from (cores/esp32/esp32-hal-i2c.c, function i2cProcQueue) I think I see some differences between Release 1.0.0 (that gives me some trouble) and Stickbreaker’s fork (that works). Basically, the flags argument in the call to esp_intr_alloc_intrstatus() is now a combination of EPS_INTR_FLAG_xxx instead of just 0 in Stickbreaker’s fork. Now the crazy part: if I modify the 1.0.0 code to pass 0 as flags again (hum… yes, I tried that…), then everything seems to work (no more error, no more crash when setting up HW timer…).

### References:

cores/esp32/esp32-hal-i2c.c
```if(!i2c->intr_handle) { // create ISR for either peripheral
        // log_i("create ISR %d",i2c->num);
        uint32_t ret = 0;
        uint32_t flags = ESP_INTR_FLAG_EDGE |  //< Edge-triggered interrupt
          ESP_INTR_FLAG_IRAM |  //< ISR can be called if cache is disabled
          ESP_INTR_FLAG_LOWMED;   //< Low and medium prio interrupts. These can be handled in C.

        log_w("MY TRACE: i2c->num=%d", i2c->num); // Trace I added….
         if(i2c->num) {
            ret = esp_intr_alloc_intrstatus(ETS_I2C_EXT1_INTR_SOURCE, flags, (uint32_t)&i2c->dev->int_status.val, 0x7FF, &i2c_isr_handler_default,i2c, &i2c->intr_handle);
        } else {
            ret = esp_intr_alloc_intrstatus(ETS_I2C_EXT0_INTR_SOURCE, flags, (uint32_t)&i2c->dev->int_status.val, 0x7FF, &i2c_isr_handler_default,i2c, &i2c->intr_handle);
        }

        if(ret!=ESP_OK) {
            log_e("install interrupt handler Failed=%d",ret);
            I2C_MUTEX_UNLOCK();
            return I2C_ERROR_MEMORY;
        }
    }
```
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_err.h
```#define ESP_ERR_NOT_FOUND           0x105   /*!< Requested resource not found */```

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/intr_alloc.html
```* @return ESP_ERR_INVALID_ARG if the combination of arguments is invalid.
*         ESP_ERR_NOT_FOUND No free interrupt found with the specified flags
*         ESP_OK otherwise
*/
esp_err_t esp_intr_alloc_intrstatus(int source, int flags, uint32_t intrstatusreg, uint32_t intrstatusmask, intr_handler_t handler, void *arg, intr_handle_t *ret_handle);
```
https://github.com/espressif/esp-idf/blob/3276a13/components/esp32/include/esp_intr_alloc.h
```
* @param flags An ORred mask of the ESP_INTR_FLAG_* defines. These restrict the
*               choice of interrupts that this routine can choose from. If this value
*               is 0, it will default to allocating a non-shared interrupt of level
*               1, 2 or 3. If this is ESP_INTR_FLAG_SHARED, it will allocate a shared
*               interrupt of level 1. Setting ESP_INTR_FLAG_INTRDISABLED will return 
 *               from this function with the interrupt disabled.
```

### Traces (esp32 + app):
```
[D>][    0.03][ZSER][ZSeries.h:540] setup BEGIN 0
[D_][    0.03][ZSER][ZSeries.h:394] screenPower() True
[I_][    0.04][SYST][System.h:267] scan()
[D>][    0.04][I2C][System.h:89] scan BEGIN 0
[W][esp32-hal-i2c.c:974] i2cProcQueue(): MY TRACE: i2c->num=0
[IV][    0.06][I2C] Name = I2CA
[IV][    0.06][I2C] Expected = 0x19 0x34 0x76
[IV][    0.06][I2C] Found = 0x19 0x34 0x76
[IV][    0.06][I2C] Missing =
[IV][    0.06][I2C] hasError = False
[D<][    0.07][I2C][System.h:89] scan END   29
[D>][    0.07][I2C][System.h:89] scan BEGIN 0
[W][esp32-hal-i2c.c:974] i2cProcQueue(): MY TRACE: i2c->num=1
[E][esp32-hal-i2c.c:982] i2cProcQueue(): install interrupt handler Failed=261
[W][esp32-hal-i2c.c:974] i2cProcQueue(): MY TRACE: i2c->num=1
[E][esp32-hal-i2c.c:982] i2cProcQueue(): install interrupt handler Failed=261

// … LOTS of errors 

[W][esp32-hal-i2c.c:974] i2cProcQueue(): MY TRACE: i2c->num=1
[E][esp32-hal-i2c.c:982] i2cProcQueue(): install interrupt handler Failed=261
[IV][    1.65][I2C] Name = I2CB
[IV][    1.66][I2C] Expected = 0x40
[IV][    1.66][I2C] Found =
[IV][    1.66][I2C] Missing = 0x40
[IV][    1.67][I2C] hasError = True
[D<][    1.67][I2C][System.h:89] scan END   1597
[E_][    1.67][SENS][SystemState.h:70] setError I2CB
[D>][    1.68][BSL][Msp.h:457] setup BEGIN 0
[IV][    1.70][MSPA] CRC_result = 57411
[W][esp32-hal-i2c.c:974] i2cProcQueue(): MY TRACE: i2c->num=1
[E][esp32-hal-i2c.c:982] i2cProcQueue(): install interrupt handler Failed=261
[W][esp32-hal-i2c.c:974] i2cProcQueue(): MY TRACE: i2c->num=1
[E][esp32-hal-i2c.c:982] i2cProcQueue(): install interrupt handler Failed=261
[IV][    1.72][MSP] bootRunning = False
[IV][    1.72][MSP] appRunning = False
[E_][    1.72][MSP][Msp.h:488] Could not figure out what MSP is currently running
[IV][    1.73][MSP] TRANSITION = Unknown -> Error
[D<][    1.74][BSL][Msp.h:457] setup END   59
[I_][    1.74][EPRM][Eeprom.h:138] save()
[D>][    1.78][MAX11613][MAX11613.h:265] setup BEGIN 0
[D<][    1.79][MAX11613][MAX11613.h:265] setup END   9
[D>][    1.79][HEAD][Head.h:645] setup BEGIN 0
[D>][    1.79][DSP][Display.h:253] setup BEGIN 0
[D<][    1.79][DSP][Display.h:253] setup END   1
[D<][    1.79][HEAD][Head.h:645] setup END   4
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40180f03  PS      : 0x00060e30  A0      : 0x800f6c06  A1      : 0x3ffb3ed0
A2      : 0x00000000  A3      : 0x0000060e  A4      : 0x40081df4  A5      : 0x00000001
A6      : 0x3ffd2730  A7      : 0x00000000  A8      : 0x800feea0  A9      : 0x3ffb3e80
A10     : 0x00000105  A11     : 0x0000060e  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x40081df4  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40180f03:0x3ffb3ed0 0x400f6c03:0x3ffb3ef0 0x400eab45:0x3ffb3f10 0x400eabb2:0x3ffb3f80 0x400f026b:0x3ffb3fa0

Rebooting...

0x40180f03: esp_intr_get_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 772
0x40081df4: __timerISR at C:\Users\adrien\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-timer.c line 173
0x40081df4: __timerISR at C:\Users\adrien\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-timer.c line 173
0x40180f03: esp_intr_get_cpu at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/intr_alloc.c line 772
0x400f6c03: timerAttachInterrupt at C:\Users\adrien\Documents\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-timer.c line 173
```

stickbreaker, 2018-09-18:
I'll have to do some testing, I'll get back with you tomorrow.

AdrienFelon, 2018-09-19:
Huh. Thanks a lot for trying and validating this scenario. I'm working on the finding the minimal sketch that reproes this.

AdrienFelon, 2018-09-19:
I think I found root cause... and it's my own fault. Basically, I had to redefine app_main to extend the stack size (defaults to 8K in arduinos'port), and it looks like I pinned the task to the wrong core...  Wrong as in 0. For reference below, the minimal sketch that reproes the problem for me. Thanks a lot for your answer, and all the work you put in I2C BTW...

```c
#include <Wire.h>
#include <WiFi.h>

#define MSP_SDA               25
#define MSP_SCL               26
#define SENSOR_SDA            22
#define SENSOR_SCL            23
  
void scan(TwoWire & bus)
{
  Serial.println("scan");
  for (uint8_t i = 0; i <= 0x7F; ++i) {
    bus.beginTransmission(i);
    if (0 == bus.endTransmission(true)) // If a device exist on bus, it will reply with 0
    {
      Serial.print("  Found ");
      Serial.println(i);
    }       
  }
}

void setup() {
  Serial.begin(115200);
  Serial.println("Started");

      WiFi.persistent(false);
      WiFi.mode(WIFI_OFF);
      WiFi.setAutoConnect(false); // Must be called after WiFi.mode()
//      WiFi.macAddress(Mac);

  
  Wire.begin(SENSOR_SDA, SENSOR_SCL, 400000);
  Wire1.begin(MSP_SDA, MSP_SCL, 400000);

  scan(Wire);
  scan(Wire1);
}

void loop() {

}

// We redefine the boot logic as originally defined in
// XXX\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void myTask(void *pvParameters)
{
  setup();
  for(;;) {
    micros(); //update overflow
    loop();
  }
}

extern "C" void app_main()
{
    initArduino();
    // Note: we allocate ourselve 
    xTaskCreatePinnedToCore(myTask, "myTask", 8192*2, NULL, 1, NULL, 0); // 0 seems to be the WRONG CORE!...
}
```

AdrienFelon, 2018-09-20:
I deleted my earlier post where I thought it was my code at fault, as I am now able to repro the issue with the sketch below that uses Arduino 1.8.5. It looks like initializing 2 HW timers and 2 Wire is required for the problem to happen. I'm wondering if I am just trying something not supported...

```
#include <Wire.h>
#include <WiFi.h>

#define MSP_SDA               25
#define MSP_SCL               26
#define SENSOR_SDA            22
#define SENSOR_SCL            23

void scan(TwoWire & bus)
{
  Serial.println("scan");
  for (uint8_t i = 0; i <= 0x7F; ++i) {
    bus.beginTransmission(i);
    if (0 == bus.endTransmission(true)) // If a device exist on bus, it will reply with 0
    {
      Serial.print("  Found ");
      Serial.println(i);
    }       
  }
}

void IRAM_ATTR ISR_Timer_1Hz() { /* do something */}
void IRAM_ATTR ISR_Timer_50Hz() { /* do something else */}

hw_timer_t * _isrTimer1Hz;
hw_timer_t * _isrTimer50Hz;

void setup2Timers()
{
  Serial.println("setup2Timers");
  _isrTimer1Hz  = timerBegin(0, 80, true);
  _isrTimer50Hz = timerBegin(1, 80, true);
  timerAttachInterrupt(_isrTimer1Hz,  ISR_Timer_1Hz,  true);
  timerAttachInterrupt(_isrTimer50Hz, ISR_Timer_50Hz, true);
}

void setup2Wire()
{
  Serial.println("setup2Wire");
  Wire.begin(SENSOR_SDA, SENSOR_SCL, 400000);
  Wire1.begin(MSP_SDA, MSP_SCL, 400000);
}

uint32_t nexpoll = 0;
void setup() {
  Serial.begin(115200);

  setup2Timers();
  setup2Wire();
  scan(Wire);
  scan(Wire1);
}

void loop() {
}
```

Yields the following output

```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
[... 244 lines cut ...]
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
[E][esp32-hal-i2c.c:981] i2cProcQueue(): install interrupt handler Failed=261
```

AdrienFelon, 2018-09-20:
Another version of the repro sketch with #define to illustrate crash in HW timer as well.

```c
#include <Wire.h>
#include <WiFi.h>

// Only define one of the 2 REPRO_X macro at a time
#define REPRO_FAILURE_IN_I2CPROCQUEUE
//#define REPRO_CRASH_IN_TIMERATTACHINTERRUPT

void IRAM_ATTR ISR_Timer_1() { /* do something */}
void IRAM_ATTR ISR_Timer_2() { /* do something */}

hw_timer_t * _isrTimer1;
hw_timer_t * _isrTimer2;

void setup1HwTimer()
{
  Serial.println("setup1HwTimer");
  _isrTimer1  = timerBegin(0, 80, true);
  timerAttachInterrupt(_isrTimer1,  ISR_Timer_1,  true);
}

void setup2HwTimers()
{
  Serial.println("setup2HwTimers");
  _isrTimer1  = timerBegin(0, 80, true);
  _isrTimer2  = timerBegin(1, 80, true);
  timerAttachInterrupt(_isrTimer1,  ISR_Timer_1,  true);
  timerAttachInterrupt(_isrTimer2,  ISR_Timer_2,  true);
}

void scan(TwoWire & bus)
{
  Serial.println("scan");
  for (uint8_t i = 0; i <= 0x7F; ++i) {
    bus.beginTransmission(i);
    if (0 == bus.endTransmission(true)) // If a device exist on bus, it will reply with 0
    {
      Serial.print("  Found ");
      Serial.println(i);
    }       
  }
}

void setup2Wire()
{
  Serial.println("setup2Wire");
  Wire.begin(22, 23, 400000);
  Wire1.begin(25, 26, 400000);
}

void setup() {
  Serial.begin(115200);

#ifdef REPRO_FAILURE_IN_I2CPROCQUEUE
  Serial.println("REPRO_FAILURE_IN_I2CPROCQUEUE");
  setup1HwTimer();
  setup2Wire();
  scan(Wire);
  scan(Wire1);
#endif

#ifdef REPRO_CRASH_IN_TIMERATTACHINTERRUPT
  Serial.println("REPRO_CRASH_IN_TIMERATTACHINTERRUPT");
  setup2Wire();
  scan(Wire);
  scan(Wire1);
  setup2HwTimers(); // -> Crashes while setting up the second timer
#endif

}

void loop() { }
```

stickbreaker, 2018-09-20:
@AdrienFelon from esp-idf/esp_intr_alloc.h:
> ESP_ERR_INVALID_ARG if the combination of arguments is invalid.
          ESP_ERR_NOT_FOUND No free interrupt found with the specified flags
          ESP_OK otherwise

It looks like there are no available interrupts?

in `esp32-hal-i2c.c` try changing the interrupt allocation flags from:
```c++
uint32_t flags = ESP_INTR_FLAG_EDGE |  //< Edge-triggered interrupt
          ESP_INTR_FLAG_IRAM |  //< ISR can be called if cache is disabled
          ESP_INTR_FLAG_LOWMED;   //< Low and medium prio interrupts. These can be handled in C.
```
to:
```c++
uint32_t flags =  ESP_INTR_FLAG_IRAM |  //< ISR can be called if cache is disabled
          ESP_INTR_FLAG_LOWMED |   //< Low and medium prio interrupts. These can be handled in C.
          ESP_INTR_FLAG_SHARED; // Share one interrupt, use i2c->dev->int_status.val to select 
```

Chuck.

AdrienFelon, 2018-09-20:
This seems to work. The repro sketch posted above does not repro any problem with these changes: no more "install interrupt handler Failed=261" and no more crashes in timerAttachInterrupt(). I also tested my app: both i2c buses seem operating just fine (I have yet to test HW timers).

For what it's worth, esp32-hal-timer.c seems to be using the exact same flags than the ones currently used by I2C:
```c
            esp_intr_alloc(intr_source, (int)(ESP_INTR_FLAG_IRAM|ESP_INTR_FLAG_LOWMED|ESP_INTR_FLAG_EDGE), __timerISR, NULL, &intr_handle);
```

closed 2018-11-30 by AdrienFelon