espressif/arduino-esp32 #1870: TG1WDT_SYS_RESET crashes while using Wifi calls

theDontKnowGuy, 2018-09-18:
Migrating ESP8266 project to ESP32. 

Created some layered network access functions to ease various calls.

Getting unstable network response (sometimes response received, sometimes not), getting different header response than in ESP8266 when using WiFiClient.

full code is in this branch: 
https://github.com/theDontKnowGuy/myRFDevicesHub/tree/ESP32-migration

19:29:30.565 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:29:30.602 -> configsip: 0, SPIWP:0xee
19:29:30.602 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:29:30.602 -> mode:DIO, clock div:1
19:29:30.602 -> load:0x3fff0018,len:4
19:29:30.602 -> load:0x3fff001c,len:952
19:29:30.602 -> load:0x40078000,len:6084
19:29:30.602 -> load:0x40080000,len:7936
19:29:30.602 -> entry 0x40080310

Board:							ESP32 Dev Module
IDE name:							Arduino IDE
Flash Frequency:					tried both
PSRAM enabled:                                         tried both
Upload Speed:				          921600
Computer OS:                                            MacOS

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

copercini, 2018-09-19:
`Stack canary watchpoint triggered` means that your task stack is not enough

theDontKnowGuy, 2018-09-19:
Thank for copercini for your response!
Is this the case even if I am not using task scheduler at all ? 
Does it basically mean that the regular stack got full ?
thanks again.

copercini, 2018-09-19:
yes, the stack of main task behind setup() and loop() is full

for test you can increase from 8192 to something bigger here https://github.com/espressif/arduino-esp32/blob/a59eafbc9dfa3ce818c110f996eebf68d755be24/cores/esp32/main.cpp#L24

but 8192 is already a large number, so this may a signal to you review your code structure ;)

closed 2019-08-15 by stale[bot]