espressif/arduino-esp32 #1871: http.POST() Failure to send large files

HelloDB, 2018-09-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			2018-09-19
IDE name:							Platform.io
Flash Frequency:					240Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10
### Description:
http.POST() Failure to send large files，For example, 30kbyte pictures.
When I narrowed the image to 10KB, I successfully sent it.

### Sketch: 
```cpp

if (wifiMulti.run() == WL_CONNECTED) {

	String url = gFace_url_add + "?access_token=" + gFace_TK;
	
	char *output = new char[b64_enc_len(image_size)];
	size_t len = b64_encode(output, (char *)image, image_size);
	// Serial.write((uint8_t *)output, len);
	String data = "{\"group_id\":\"" + gSN + "\",\"user_id\":\"user_001\",\"user_info\":\"user_test\"," + // 图像信息
				  "\"image_type\":\"BASE64\",\"image\":\"" + output + "\"," +							  // 图像数据
				  "\"quality_control\":\"NORMAL\"," +													  // 图像质量要求，这里设置为一般
				  "\"liveness_control\":\"NORMAL\"" +													  // 活体检测控制，这是设置为一般
				  "}";

	// delete[] output;

	LOGln(url);
	// LOGln(data);
	HTTPClient http;
	http.begin(url);
	LOGln("http begin...");
	http.addHeader("Content-Type", "application/json");
	int httpCode = http.POST(data);
	if (httpCode > 0) {
		LOGf("POST... code: %d\n", httpCode);
		if (httpCode == HTTP_CODE_OK) {
			String payload = http.getString();
			Serial.println(payload);
		}
	} else {
		LOGf("POST... failed, error: %s\n", http.errorToString(httpCode).c_str());
	}
	http.end();
}

```

### Debug Messages:
```
[Fn:face_add        :230 ]:http begin...
[Fn:face_add        :248 ]:POST... failed, error: send payload failed 

```

mikkel75, 2019-04-01:
Any workaround or fix for this?

mikkel75, 2019-04-01:
I "hacked" the HTTPClient.cpp sendRequest() send Payload part to the following and it works fine now with eg. images of 250+k:

```
    // send Payload if needed
    if(payload && size > 0) {
        int cIndex;
        for (cIndex = 0; cIndex < size; cIndex = cIndex + 1000) {
          int left = size - cIndex;
          if ( left > 1000 ) {
            if(_client->write(&payload[cIndex], 1000) != 1000) {
              return returnError(HTTPC_ERROR_SEND_PAYLOAD_FAILED);
            }
          }
          else { // write last bit
            if(_client->write(&payload[cIndex], left) != left) {
              return returnError(HTTPC_ERROR_SEND_PAYLOAD_FAILED);
            }
          }
        }
    }

```

Please make a proper fix for this in sources.....

mipoat, 2019-04-03:
I just came across the very same issue:
Sending 32k buffers via http.POST() worked most of the time, but as soon as I tried to send something bigger it failed with "send payload failed".
Applying the hack above solved my issue.

mikkel75, 2019-05-06:
Hi,

Why has this not been fixed in master branch yet? Who should I write to...?

Thanks

PeterKillerio, 2019-06-16:
Had similar issue with esp32-cam sending img post requests, mikkel75's solution fixed it for me. Thank you so much

closed 2019-08-29 by stale[bot]

NavadeepGaneshU, 2021-10-05:
I tried with the "hack" using the same snippet suggested and doesn't seem to work out for me. Any suggestions why?
My application is to send a ~320kB file through http.POST( )