espressif/arduino-esp32 #1874: ESP32 mDNS unresponsive after few seconds.

nomanshahsyed, 2018-09-20:
### Hardware:
Board:							Adafruit ESP32 Feather
update date:			                       20-09-2018
IDE name:					       Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600
Computer OS:                                          Windows 7

### Description:
I am making an HTTP server on ESP32, but the first step is connecting to WiFi. Then esp starts mDNS server so that the IP can be read which is working fine. This mDNS server replies IP perfectly. But the problem is this mDNS server becomes unresponsive if it is not requested the IP. I have put a delay using millis() also so that if the server is not requested 20 seconds the ESP restarts. Which also works fine for some minutes then ESP again becomes unresponsive.

Code steps are below:

1. esp looks for stored SSID and Password, if found esp connects and gets IP.
2. mDNS server starts and waits for client connection for 20 seconds. Otherwise retarts.
3. using .local the esp sends the IP.

**but after these steps the system becomes unresponsive if there is no interaction.**

4. app sends a request which says that ip is read and then the mDNS function is not executed instead http function is handled. This is also working fine.

 

//Change the code below by your sketch

#include <ArduinoJson.h>
#include <WiFiClientSecure.h>
#include "WiFi.h"
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <WiFiClient.h>
#include <ESP32WebServer.h>
#include <Preferences.h>

ESP32WebServer HTTP(80);
WiFiServer server(80);

unsigned long previousMillis;

IPAddress ip;
String ipStr, req, s, wifiSSID, wifiPassword, ipRead;
const int wifiButton = 14;

Preferences preferences;

void ssidPass()
{
  wifiSSID = preferences.getString("ssid");
  wifiPassword = preferences.getString("password");
  Serial.print("Stored SSID: ");
  Serial.println(wifiSSID);
  Serial.print("Stored Pass: ");
  Serial.println(wifiPassword);

  WiFi.begin(wifiSSID.c_str(), wifiPassword.c_str());

  if (WiFi.status() != WL_CONNECTED)
  {
    for (int i = 0; i < 10; i++)
    {
      delay(200);
      Serial.print(".");
    }
  }

  ip = WiFi.localIP();
  ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
  Serial.println(ipStr);
}

void wifiConfig()
{

  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();

  while (!WiFi.smartConfigDone())
  {
    delay(500);
    Serial.print(".");
  }

  preferences.putString("ssid", WiFi.SSID().c_str());
  preferences.putString("password", WiFi.psk().c_str());

  Serial.println(preferences.getString("ssid"));
  Serial.println(preferences.getString("password"));
}
```
void wifi()
{
  Serial.print("Connecting");

  wifiSSID = preferences.getString("ssid");
  wifiPassword = preferences.getString("password");

  WiFi.begin(wifiSSID.c_str(), wifiPassword.c_str());

  int t = 0;
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
    t += 1;
    if (t == 20)
    {
      ESP.restart();
    }
  }
  ip = WiFi.localIP();
  ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
  Serial.println(ipStr);
}

void mdnsSetup()
{

  if (!MDNS.begin("name"))
  {
    Serial.println("Error setting up MDNS responder!");
    while (1)
    {
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");

  server.begin();

  Serial.println("TCP server started");

  MDNS.addService("_http", "_tcp", 80);  **// Without _ the mDNS was not working**
}
```
void mdns()
{

  WiFiClient client = server.available();

  unsigned long currentMillis = millis();

  if ((currentMillis - previousMillis) > 20000)
  {

    Serial.println("Restarting System");
    delay(500);
    ESP.restart();
  }

  if (!client)
  {
    return;
    delay(10);
  }

  while (!client.available())
  {
  }
  req = client.readStringUntil('\r');

  while (client.available())
  {
    size_t a = client.available();
    uint8_t buf[a];
    client.readBytes(buf, a);
  }
  int addr_start = req.indexOf(' ');
  int addr_end = req.indexOf(' ', addr_start + 1);
  req = req.substring(addr_start + 1, addr_end);
  Serial.print("Request: ");
  Serial.println(req);

  client.flush();

  if (req == "/")
  {
    s = "HTTP/1.1 200 OK\r\nContent-Type: text/plain\r\nContent-Length: 10\r\nAccess-Control-Allow-Origin: *\r\nAccess-Control-Allow-Headers: *\r\n\r\n" + ipStr + "";

**//These headers are there for mobile application. Without them, the reply is not sent.**

    Serial.println("Sending 200");
  }  else
  {
    s = "HTTP/1.1 404 Not Found\r\n\r\n";
    Serial.println("Sending 404");
    preferences.putString("ipRead", "No");
  }

  client.print(s);
}
```
void http()
{
  if (WiFi.waitForConnectResult() == WL_CONNECTED)
  {
    Serial.println("starting http");
   // Here are some HTTP requests.
    HTTP.begin();
  }
  else
  {
    Serial.printf("WiFi Failed\n");
    while (1)
      delay(2000);
  }
}
```
void setup()
{
  Serial.begin(115200);

  preferences.begin("wifi", false);

  previousMillis = millis();
  Serial.println(previousMillis);
  ssidPass();

  if (WiFi.status() != WL_CONNECTED)
  {

    Serial.println("Wifi not connected. Starting Smartconfig.");

    preferences.putString("ipRead", "No");

    wifiConfig();

    wifi();
  }

  ipRead = preferences.getString("ipRead");

  Serial.println(ipRead);

  if (ipRead == "Yes")
  {
    Serial.println("Ip is already stored.");
    http();
  }
  else
  {
    mdnsSetup();
  }
  Serial.println("Have we read IP already");

  Serial.println(ipRead);
}
```
void loop()
{

  ipRead = preferences.getString("ipRead");

  if (ipRead == "Yes")
  {
    HTTP.handleClient();
  }
  else
  {
    mdns();
  }
}

```

### Debug Messages:
```

**After that unresponsive phase, I hard reset and the outupt is below.**

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
```

nomanshahsyed, 2018-09-20:
**once or twice when ESP restarts, it cant connect to WiFi even though the ssid and password are already read.**

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
19
Stored SSID: asdfsdf 
Stored Pass: afsdfsaf
..........0.0.0.0
Wifi not connected. Starting Smartconfig.

closed 2019-08-15 by stale[bot]

Yurik72, 2020-12-14:
Unfortunatelly I meе with the same issue.  But found a solution, removing power save mode (is by deafault) solves a problem
`
esp_wifi_set_ps(WIFI_PS_NONE);
`
I suppose that is not software issue, probasbly is something related to hardware or mDns is not properly supported in the modem power save mode (which cab be understandable) , so any comment from espressif or documentation changing appreciated

me-no-dev, 2020-12-18:
When sleep is enabled, the radio wakes up every 100ms to check for packets. Since UDP does not have a way to keep track of sent/received data, it is normal to lose some packets. mDNS is UDP