espressif/arduino-esp32 #1876: Close abandoned Wire and I2C issues

stickbreaker, 2018-09-20:
@everslick can you close these Issues, they are no longer relevant.
* #246 Fixed with RC1
* #784 duplicate of #118
* #1093 fixed with RC1
* #1236 Fixed with RC1, boot strapping pin problem
* #1352 Fixed PR #1767
* #1438 Fixed, user coding error
* #1558 Fixed PR #1767
* #1561 this one is just bizarre, it is abandoned and the OP was just acting silly
* #1563 Fixed PR #1767 
* #1588 fixed with PR #1767 
* #1657 fixed with PR #1717 
* #1694 fixed with PR #1717 
* #1697 Duplicate of #118
* #1725 request to change Wire.endTransmission(sendStop) parameter to byte, not accepted
* #1778 Problem with user's board/hardware

Chuck.

everslick, 2018-09-20:
Will do so the next days... Thanks for your work!

copercini, 2018-09-21:
@stickbreaker I closed all the issues on list, except the ones that still are pending the PR:

- #1352 Fixed PR #1767
- #1558 Fixed PR #1767
- #1563 Fixed PR #1767
- #1563 Fixed PR #1767
- #1588 fixed with PR #1767
- #1657 fixed with PR #1717
- #1694 fixed with PR #1717

will close after merge =)

closed 2018-09-21 by copercini

bperrybap, 2018-09-22:
In #1725  I wasn't asking to change Wire.endTransmission(sendStop) parameter to use a byte parameter.
I was saying that endTransmission(0) and endTransmission(1) should work as it does in every other Arduino core.

bperrybap, 2018-09-22:
PR created for issue #1725 : https://github.com/espressif/arduino-esp32/pull/1888