espressif/arduino-esp32 #1879: WiFi Status Not Updated on Disconnect

RijadBajramovic, 2018-09-20:
**Hardware:**
Board: Adafruit HUZZAH32
Core Installation/update date:	IDF & Arduino up to date
IDE name:	PlatformIO
Flash Frequency:	40Mhz
Upload Speed:	115200

**Description**
I have an application that uses 2 FreeRTOS tasks. One of these tasks will eventually act as a server that clients connect to.

I attempted to implement auto reconnect to WiFi. To test this I uploaded my code, ensured the ESP32 connected to WiFi,  unplugged my router from the wall, waited a few minutes, and plugged it back in.

It appears that while my router is disconnected the return value from WiFi.status() was WL_CONNECTED (3) which indicated the ESP32 thinks it's still connected to WiFi. I have waited for as long as 10 minutes and the WiFi status did not update. 

I know my WiFi didn't work because all of my other devices couldn't connect.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>

#define SSID "**********"
#define PASSWORD "**********"

TaskHandle_t otherTask;
TaskHandle_t wifiTask;

unsigned long int tempTime = 0;

void otherThread(void*ptr)
{
  while(1)
  {
    //do something else here
    delay(100);
  }

  vTaskDelete( NULL );

}

void wifiThread(void *ptr)
{
  WiFi.begin(SSID, PASSWORD);

  Serial.print("WiFi status: "); Serial.println(WiFi.status());

  while (WiFi.status() != WL_CONNECTED) 
  {
    delay(300);
    Serial.println("Connecting to WiFi....");
  }

  Serial.print("IP: ");Serial.println(WiFi.localIP());

  while(1)
  {
    if((WiFi.status() != WL_CONNECTED))
    {
      Serial.println("WiFi Disconnected");
      WiFi.disconnect();

      delay(100);

      WiFi.begin(SSID,PASSWORD);

      while(WiFi.status() != WL_CONNECTED)
      {
        delay(300);
        Serial.println("Attempting to reconnect to WiFi");
      }
    }

    else
    {
      unsigned long int currentTime = millis();

      if((currentTime - tempTime) > 700) //non blocking print every 700ms
      {
        Serial.print("WiFi status: "); Serial.println(WiFi.status());
        tempTime = currentTime;
      }
    }
  }

  vTaskDelete( NULL );
}

void setup() 
{
  Serial.begin(115200);
  
  xTaskCreatePinnedToCore(
    otherThread,           
    "otherThread",       
    20000,                    
    NULL,                     
    3,                       
    &otherTask,           
    0);
  
    xTaskCreatePinnedToCore(
    wifiThread,          
    "wifiThread",        
    25000,                    
    NULL,                     
    3,                        
    &wifiTask,             
    1);
}

void loop()
{

}
```

karthikram05, 2019-04-25:
Hi, could you figure out as to why this problem occurs?

closed 2019-08-15 by stale[bot]

1technophile, 2019-08-28:
This problem seems to depend on the router used.
Here is a discussion about a user encountering a similir issue with an Apple Timecapsule 5.gen A1470. Firmware 7.9.1 

https://community.openmqttgateway.com/t/esp32-doesnt-stay-online/570/56?u=1technophile