espressif/arduino-esp32 #1881: Tmr Svc did not rest watchdog

dorarosi, 2018-09-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			28/jul/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          ?no?
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
Hello I am using an ESP 32 as an CAN Gateway which should be configurable over wifi. I am also using OTA. After adding the wifi functions I get the watchdog error. I have read in another [Issue](https://github.com/espressif/esp-idf/issues/666) that the Stacksize could be too small, but i don't know how to change it in Arduino IDE.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

CelliesProjects, 2018-09-22:
To increase the stack size for a task change
```
  xTaskCreatePinnedToCore(serverTask,"serverTask",10000,NULL,3,NULL,1);
````
to for example:
```
  xTaskCreatePinnedToCore(serverTask,"serverTask",20000,NULL,3,NULL,1);
```
But the error indicates the idle task is getting not enough time.

lbernstone, 2018-09-22:
This is most likely a problem with your 3rd party library which you will need to take up with the author.  However, you can help them out by putting debug statements in your functions to show how much time is spent is spent in the various processes.  Your infinite loops need to have some yields/delays in them to allow other processes to get some cpu time.  The OTA handler only needs to run about every 500ms.

dorarosi, 2018-09-24:
@CelliesProjects i know, but i don't know where the Tmr Svc is called.

@lbernstone 
> The OTA handler only needs to run about every 500ms.

changing this loop to 500ms didn't help. Also this loop runs on core 1 and the Tmr Svc on core 0, where wifi is running. 
>This is most likely a problem with your 3rd party library

This is no 3rd party library. The eps32Msp2515.h is the only 3rd Party library which is accessed in the isr's and trigCan 1&2 which are running on core 1.

> Your infinite loops need to have some yields/delays in them to allow other processes to get some cpu time.

trigCan 1&2 are halted at `xEventGroupWaitBits(eg,CAN_2_BIT,pdTRUE,pdTRUE,portMAX_DELAY);` and then they don't consume any cpu time.

Now this message
```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: Tmr Svc
CPU 1: IDLE
```
alternates with this one
```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi
CPU 1: IDLE
```

So I Think this is a problem with wifi

dorarosi, 2018-09-27:
So I updated the Core and it worked fine for about 2 days. Now the Tmr Svc comes after around 1 minute, even when wifi is commented out. I eveb changed the ESP and the MCPs and i didn't worked. Guys I'm helpless.

closed 2019-08-15 by stale[bot]

Padifu03, 2022-11-09:
Did you find any solution? I am having same the same error related with tmr svc : ebug exception reason: Stack canary watchpoint triggered (Tmr SVC). 
I have found that there is a problem with service timer size stack but it was fixed with a 4Mb size in an arduino esp32 library update but still appears the tmr svc error in my case.

Padifu03, 2022-11-09:
> Did you find any solution? I am having same the same error related with tmr svc : Debug exception reason: Stack canary watchpoint triggered (Tmr SVC). I have found that there is a problem with service timer size stack but it was fixed with a 4Mb size in an arduino esp32 library update but still appears the tmr svc error in my case.

I have found a solution. in my case. I check my ESP32 SoC, which is WROOM-32U revision 1 and I found that there are some bugs due to version. So, I just erase flash and flash the Arduino IDE ESP32 WiFi Client sample and probably It has changed a register related to core 1 and WiFi and now the SoC works well with WiFi.
I am not really sure about which exactly changed but it works. If anyone have idea of what changed please answer this post or send me an email.