espressif/arduino-esp32 #1884: Task stack size... how much?

Darkhub, 2018-09-21:
I am experimenting with multitasking on ESP32 using Arduino IDE and I have a question about stack size when creating tasks with xTaskCreate or xTaskCreatePinnedToCore functions.

Is there a way to calculate how much stack is needed?

According to https://www.freertos.org/uxTaskGetStackHighWaterMark.html I have tried to determine the maximum stack size effectively used by my task and noticed that if I lower the value of the stack size when I create the task, then the value returned by uxTaskGetStackHighWaterMark lowers too.
As an example I created a task with stack size = 4000, max stack used was 3700.
So I edited the progam and lowered the stack size to 3800, started the program and uxTaskGetStackHighWaterMark reported 3600 max stack used.
Repeated the procedure and everytime the used stack size was Always lower according to the stack size parameter of xTaskCreate function.

So, how do I determine the minimum required stack size? (If I am correct it's 1024)
Do I have to lower it and check until I get a stack overflow error?

Thanks in advance for support.

lbernstone, 2018-09-21:
This is an ESP-IDF issue, not arduino. RTFM at https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/freertos.html and https://www.freertos.org/a00111.html.

chegewara, 2018-09-21:
`uxTaskGetStackHighWaterMark` shows how much stack was never used not the value of used stack. It should have to report as close to 0 as its possible (with some margin).

Darkhub, 2018-09-22:
> `uxTaskGetStackHighWaterMark` shows how much stack was never used not the value of used stack. It should have to report as close as possible to 0 (with some margin).

I see, my mistake then, many thanks.
So considering it in the right way it works as expected, stack used is almost 200 and lowering the maximum stack lowers the unused size. This is correct then, nice.

closed 2018-09-22 by Darkhub

jandolina, 2022-10-06:
My app is running LoRa and OTA on a separate core from my main logic. Do any of you know if the OTA update size will affect the stack size required?