espressif/arduino-esp32 #1887: Error during compilation with newest arduino-esp32 and AsyncTCP

smacyas, 2018-09-22:
### Hardware:
Board:							ESP32 Dev Module (WROOM32)
Core Installation/update date:			Sep 21, 2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
Describe your problem here

Error during compilation with AsyncTCP + ESPAsyncWebServer
https://github.com/me-no-dev/AsyncTCP/issues/21

copercini, 2018-09-22:
there is a idf-update branch on https://github.com/me-no-dev/AsyncTCP/tree/idf-update, try this one ;)

smacyas, 2018-09-23:
Thanks, with this one compilation is ok!

mvadu, 2018-11-03:
Thanks!! I can confirm that `idf-update` branch does indeed compile. What is your plan forward to merge this with master branch?

copercini, 2018-11-03:
It depends of @me-no-dev

eziosoma, 2019-01-12:
I'm using PlatformIo and I am experiencing he same issue. What I have to do for updating to the idf-update branch on https://github.com/me-no-dev/AsyncTCP/tree/idf-update in PlatformIO?
Many thanks, Ezio

me-no-dev, 2019-01-12:
idf-update branch is merged and version is updated. you will not have issues

eziosoma, 2019-01-12:
How can I Update to the new version? PlatformIO seems not recognize that there is a new (merged) library version.

eziosoma, 2019-01-12:
I have also tryied to uninstall then reinstall the ESP Async WebServer library, but without success.

eziosoma, 2019-01-12:
What I have to do for having Platformio seeing the update version? i have tried with the normal commands but it does not show the existence of an update for the ESP Asyn Webserver library and AsyncTCP library.
The last AsyncTCP version shown is 1.0.2 instead of 1.0.3

eziosoma, 2019-01-12:
I manually downloaded the library and uploaded in PlatformIO and this did the trick.
I'm just wondering if this was the right way of doing this upgrade...

me-no-dev, 2019-01-12:
maybe pio needs some time to grab the update :)

bertmelis, 2019-01-12:
'pio lib update' in the platformio terminal.

bertmelis, 2019-01-12:
or delete the .piolibdeps folder

eziosoma, 2019-01-12:
i have update by hand the library by substituting files in the directory /home/<user>/.platformio/lib/AsyncTCP_ID1826
Now it compiles without errors, but the ESP32 never connect to wifi.

The piece of code below never exit:

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

Any idea?

Thanks.

bertmelis, 2019-01-13:
You should ask this question on the Platformio community forum.
Else: run `pio update` and `pio lib update`.

eziosoma, 2019-01-13:
It seems an hardware problem because if I detach the ESP from the breadboard it runs.
If I detach some wire from the breadboard it runs.
Very strange behavior, but not seems tied to Software.

borch84, 2019-01-13:
I would like to report that I am having same issue with AsyncTCP library. I am running arduino ide 1.8.8 and if I update to esp core 1.0.1 I get compilation errors. If I rollback to esp 1.0.0 it compiles successfully.

borch84, 2019-01-13:
hi @bertmelis I forgot to update AsyncTCP library. Now I was able to compile with ESP32 core version 1.0.1, Arduino IDE 1.8.8 and AsyncTCP library version 1.0.3

closed 2019-08-15 by stale[bot]

froussel, 2020-04-07:
Hello, I'm trying to compile
https://robotzero.one/esp32-cam-rc-car/
But I have the same problem as above
libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory
AsyncTCP is installed

lbernstone, 2020-04-07:
Do not hijack issues.
https://github.com/me-no-dev/ESPAsyncWebServer#espasyncwebserver

froussel, 2020-04-07:
Found: my ino files are in ../arduino/sources/
Do not declare the ../arduino/sources  in "preferences" but ../arduino

now :
Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:210:5: error: 'AsyncClient' does not name a type
I have ide and esp32  : all update made

lbernstone, 2020-04-07:
This has nothing to do with the OP issue.  Open a new issue if you want help.