espressif/arduino-esp32 #1889: SPI.begin does not set pinMode of SS as output

pepe79, 2018-09-22:
Reading the original arduino documention of SPI.begin (https://www.arduino.cc/en/Reference/SPIBegin) they say that the spi bus is configured setting the pinMode of SS to OUTPUT. 
The SPI-implementation of arduino-esp32 doesn't do this.

I understand that this is totally unnecessary in Master-mode, but it introduces a major incompatibility to libs that are relying on that and where the SS pin is not explicitly set as output by the lib.

https://github.com/espressif/arduino-esp32/blob/e5ea089a7fd5cc8a96d68aae64dc3b4dcef2315e/libraries/SPI/src/SPI.cpp#L37-L68

pepe79, 2019-08-01:
Seems like noone has an opinion about that :-)

atanisoft, 2019-08-01:
@pepe79 submit a PR that implements this and it can be evaluated for inclusion.

pepe79, 2019-08-09:
I am really unsure, if I had a misconception or if this is really a bug and noone has stumbled upon this problem. So my intention was to get some more opinions. In the meantime I also found another code line in https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino
where the problem is solved manually by setting SS in OUTPUT mode, but missing setting it to a HIGH value:
    
    //set up slave select pins as outputs as the Arduino API
    //doesn't handle automatically pulling SS low
    pinMode(5, OUTPUT); //VSPI SS
    pinMode(15, OUTPUT); //HSPI SS

I think that SPI.begin should set SS as OUTPUT and set SS to HIGH, as stated in https://www.arduino.cc/en/Reference/SPIBegin 
If I get some positive response, I would be happy to create a PR for that.

atanisoft, 2019-08-09:
@pepe79 I agree from the description of that Arduino spec it does seem that the SS HIGH step is missing.

closed 2019-10-22 by stale[bot]