espressif/arduino-esp32 #1890: MD5 of file does not match data in flash! (But works for the code that already running)

lucasventurasc, 2018-09-23:
### Hardware:
Board:							M5Stack with ESP32D0WDQ6
Core Installation/update date:			latest
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?921600?
Computer OS:                                            ?Linux Mint?

### Description:
Describe your problem here

Hi, after upload sketchs serveral times for ESP32 module (M5 Stack), I've started to receive the error message when I'm trying to upload:

A fatal error occurred: MD5 of file does not match data in flash!

A interesting thing is if I upload the same code (same that is running inside of the ESP32) the upload is completed, when I change just a character on the code I receive that error again.

I already tried with all frequencies, all upload speed and all flash modes, I don't know what to do now.
I read several articles and issues on GitHub and most of times seems something related to "Power Issues". But I'm using the USB-C cable connected to a notebook.

After receive this error I've uploaded more than 50 times.

Even a Hello World couldn't be uploaded.

Arduino IDE LOG:

Sketch uses 1023466 bytes (78%) of program storage space. Maximum is 1310720 bytes.
Global variables use 43644 bytes (13%) of dynamic memory, leaving 284036 bytes for local variables. Maximum is 327680 bytes.
python /home/ventura/.arduino15/packages/esp32/tools/esptool/2.3.1/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /home/ventura/.arduino15/packages/esp32/hardware/esp32/1.0.0/tools/partitions/boot_app0.bin 0x1000 /home/ventura/.arduino15/packages/esp32/hardware/esp32/1.0.0/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /tmp/arduino_build_306541/mp3withnewurl.ino.bin 0x8000 /tmp/arduino_build_306541/mp3withnewurl.ino.partitions.bin 
esptool.py v2.3.1
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 41978.9 kbit/s)...
Hash of data verified.
Compressed 15072 bytes to 9733...

Writing at 0x00001000... (100 %)
Wrote 15072 bytes (9733 compressed) at 0x00001000 in 0.1 seconds (effective 931.7 kbit/s)...
Hash of data verified.
Compressed 1023600 bytes to 578016...

Writing at 0x00010000... (2 %)
Writing at 0x00014000... (5 %)
Writing at 0x00018000... (8 %)
Writing at 0x0001c000... (11 %)
Writing at 0x00020000... (13 %)
Writing at 0x00024000... (16 %)
Writing at 0x00028000... (19 %)
Writing at 0x0002c000... (22 %)
Writing at 0x00030000... (25 %)
Writing at 0x00034000... (27 %)
Writing at 0x00038000... (30 %)
Writing at 0x0003c000... (33 %)
Writing at 0x00040000... (36 %)
Writing at 0x00044000... (38 %)
Writing at 0x00048000... (41 %)
Writing at 0x0004c000... (44 %)
Writing at 0x00050000... (47 %)
Writing at 0x00054000... (50 %)
Writing at 0x00058000... (52 %)
Writing at 0x0005c000... (55 %)
Writing at 0x00060000... (58 %)
Writing at 0x00064000... (61 %)
Writing at 0x00068000... (63 %)
Writing at 0x0006c000... (66 %)
Writing at 0x00070000... (69 %)
Writing at 0x00074000... (72 %)
Writing at 0x00078000... (75 %)
Writing at 0x0007c000... (77 %)
Writing at 0x00080000... (80 %)
Writing at 0x00084000... (83 %)
Writing at 0x00088000... (86 %)
Writing at 0x0008c000... (88 %)
Writing at 0x00090000... (91 %)
Writing at 0x00094000... (94 %)
Writing at 0x00098000... (97 %)
Writing at 0x0009c000... (100 %)
Wrote 1023600 bytes (578016 compressed) at 0x00010000 in 7.8 seconds (effective 1046.9 kbit/s)...
File  md5: 90d74b744eb2a8acade99136e4d7ea55
Flash md5: fe0949ec4dd4af242ed642e2781c707d
MD5 of 0xFF is 36ed8c0cd10f63f644ac82d112d814ef

A fatal error occurred: MD5 of file does not match data in flash!
A fatal error occurred: MD5 of file does not match data in flash!

copercini, 2018-09-23:
Olá Lucas!

I usually get this due bad flashing, like bad quality or too long USB cables, or when there are more things sharing the power supply...
Another tip, try with baud 115200 instead 921600

lucasventurasc, 2018-09-24:
Hi @copercini, thank you for reply, I'm using the M5Stack original USB C Cable, which have almost 10 centimeters I think.

I already tried all disponible bounds, more than 100 times hehe, and the error message is same the always.

There is a way to reset all flash content? like factory defaults?

Thanks

copercini, 2018-09-24:
> There is a way to reset all flash content? like factory defaults?

Yes, using `esptool.py erase_flash`

lucasventurasc, 2018-09-24:
> > There is a way to reset all flash content? like factory defaults?
> 
> Yes, using `esptool.py erase_flash`

I think I've already tried that but the old application has not erased (when I've turned up the same application was loaded).

I will try it again, this approach should work for any situation?

pchidley, 2019-02-24:
Did you find a resolution?
I have a brand new M5Stack Fire that worked for about 30 minutes.
Then "unable to auto-detect flash".
Tech support said erase the flash.  (but didn't say how)
So I erased the flash.
Now I get your "A fatal error occurred: MD5 of file does not match data in flash!"

Any answers?

samguyer, 2019-03-03:
I had the same problem, and I discovered that I had my power wires backwards on part of my Arduino circuit. Once I fixed the circuit, the upload went perfectly.

pchidley, 2019-03-03:
M5Stack Fire is a preassembled board. So not the power wires.

pchidley, 2019-03-27:
Anyone?  My M5Stack Fire worked for about 20 minutes when I got it.  Nothing but errors since.

Gabriel-Gardin, 2019-04-26:
Anyone found a solution? Same problem here, but in my case i faced this error since the first time using the board...

elf128, 2019-05-16:
Got the same issue after switching versions of ESP_IDF. Dont really remember exact order.
But effectively, it was switch from some old version to 3.1.2 ( which failed, cause I didn't clean flash and NVS was wrong ).
Then I tried to fix it, but run into MD5 issue.
After switch to 3.2 the issue stays.

The fix for me, was to switch back to my old IDF with esptool v.2.3.1.
Rebuild the project. ```make flash```, now it flashed fine.
Then switching straight to v3.2.
```
make menuconfig 
make flash
```
It will rebuild everything again.
And then, flashing works fine.

Don't ask me, how IDF version is relevant, it doesn't make any sense to me, but there we are.
My guess is it has something to do with compatibility of esptool and bootloader.

chegewara, 2019-08-06:
> Got the same issue after switching versions of ESP_IDF. Dont really remember exact order.
> But effectively, it was switch from some old version to 3.1.2 ( which failed, cause I didn't clean flash and [...]

Im having the same issue with m5stack fire and esp-idf 3.1.3. I cant flash it or even erase it. 
I know its arduino board, but issue has not been solved.

pchidley, 2019-08-06:
I have never figured out what went wrong with my first M5Stack Fire.
However I eventually got it replaced and no problems with the new one.
Probably defective hardware?

atanisoft, 2019-08-06:
> Im having the same issue with m5stack fire and esp-idf 3.1.3. I cant flash it or even erase it.
> I know its arduino board, but issue has not been solved.

IDF 3.1 and 3.2 use a different for NVS IIRC. The flash erase would be the only solution when going between versions IIRC.

chegewara, 2019-08-06:
v3.1.3 has multi-page blob support, so i am assuming its the same as for v3.2.

atanisoft, 2019-08-06:
> v3.1.3 has multi-page blob support, so i am assuming its the same as for v3.2.

Scroll down to "Changes not compatible with earlier versions" https://github.com/espressif/esp-idf/releases/tag/v3.2

Unfortunately once you move up to 3.2.x you must erase flash fully to go back to 3.1.x (or earlier). The v3.1.x does support multi-page NVS but the wear leveling changes break going backward without full erase and rewrite.

chegewara, 2019-08-06:
Thanks @atanisoft i will try it.
Erasing flash its not a problem if it finally works, i want to erase flash but forunatelly i cant.

gongfan99, 2019-08-19:
I had the same issue. My fix is to reconnect the download usb cable. I do not know why it works.

chegewara, 2019-08-19:
In my case its m5stack fire issue. After some time, where it is warm enough i have problems with 2 of my m5stack. When i disconnect it and let it cool down a bit i can flash it again.

maeshimaroo, 2019-09-27:
> > There is a way to reset all flash content? like factory defaults?
> 
> Yes, using `esptool.py erase_flash`

how to use this command?

lbernstone, 2019-09-27:
https://github.com/espressif/esptool

closed 2019-12-10 by stale[bot]

lmanoarchilla, 2020-05-22:
i´am working with a naked esp8266 not a nodemcu, i can fix this problem changing my power supply conector. Sometimes i have problems up loading my programs if GPIo15 o some similar conecction are broken.

CH_PC,VCC,GPIO2 ->3,3v
GPIO15,gnd -> GND

GPIO and reset must be in mode flash mode, too you can search your power line in the protoboard,changing more up and down in the line some conection

lidorshimoni, 2020-06-17:
I had the same problem, solved it by changing the input power pin from 5v to the 3.3v pin.

procommerz, 2020-07-03:
That's a problem related to the "fake" GND pin (the one next to 5V on a typical ESP32 NodeMCU board). Although it's marked as a normal "GND" on the silk of all virtually all my esp32 boards. When this pin is connected to GND, the board will not work as expected, and if that't the only ground connection you have, your other wires will appear as not grounded on the scope!

andreacer, 2021-01-26:
In my setup this  is caused by an anomale sequence at power-up.
 I.e. some pins receive 3.3V before main cpu. This can be caused by another machine that sends data by serial before power up.

Sherif-Makii, 2021-04-22:
I fixed this problem by disconnect 5v pin while (compiling and  uploading) and connect it after :
esptool.py v3.0-dev
Serial port COM3
Connecting..
message 
it  take many times to do it but it works at the end

woodlist, 2021-06-26:
None of above written methods helped me for AI-Thinker ESP32-CAM board. The threat has been fixed accidentally, when the GPIO3 grounded (flash mode set) and the power GND wire was disconnected. It's a magic, I think.

Noisecontrol60, 2021-08-11:
How resolve this problen?

rannbaraku, 2021-12-01:
Hey, I got some problems too. i solved it with disconnect all wiring on my esp and reupload then everything works properly

moefear85, 2022-02-20:
when flash goes bad, it can still run the currently flashed program, but it is no longer erasable or rewritable. But, the stub loaded into the esp for the sake of flashing is loaded into ram, hence completes successfully. This stub is then responsible for actually programming the flash. Either it is not designed to detect flash failures, or the flash chip doesn't have a way to signal an error back to the chip/stub. Hence esptool thinks the flashing process is working and completes without errors, but nothing in the flash actually changes. The same applies if you erase the flash. It isn't actually erasing.

There's no good solution besides using a new chip. Sometimes the defect might be transient, such as bad power, hence the flash appears to work again later. Usually people will have fiddled around with various settings in the mean time, hence get the impression that the problem (thus fix) is software based.

Transcendence1, 2022-02-22:
for me the problem was due to a low voltage on io12. I connected it to another device. After disconnecting this pin, the problem was solved.
hope it helps

chrisyouritdept, 2022-03-27:
For me I was able to fix the issue using the command: 
$ esptool.py  -p /dev/tty.SLAB_USBtoUART write_flash_status --non-volatile

PranayLendave, 2022-04-17:
Face the same problem after i guess some mishandling of ESP32 cam board 
just do one thing flash your esp32, run the following command in command prompt-
esptool --port COM12 write_flash_status --non-volatile 0
(instead of COM12 write your com port where esp32 is connected) 
after running this command the issue was fixed and i was able to upload a new program again

asapella, 2022-05-09:
> Face the same problem after i guess some mishandling of ESP32 cam board just do one thing flash your esp32, run the following command in command prompt- esptool --port COM12 write_flash_status [...]

it solved it for me thank a lot

sarmiento1994, 2022-05-13:
> 

COMO LO SOLUCIONO HERMANO??

SaddamGul842, 2022-07-21:
I designed a custom board for ESP32 Wroom and it has same issue. Then I figured it out and problem solved. The issue was that I didn't pulldown the GPIO2 of ESP32 Wroom. So I pulldowm that GPIO2 using 10Kohm resistor and problem become solved.

zeidanIabderrahmane, 2022-08-24:
I had the same problem with my ESP32  Wroom, and after searching for a while i just found a solution and it worked.
Found it on this [website](https://community.m5stack.com/topic/838/a-fatal-error-occurred-md5-of-file-does-not-match-data-in-flash/5).
You just need to type this command in your terminal : 

> esptool --port <YOUR PORT> write_flash_status --non-volatile 0

jvcaloen, 2022-12-01:
I had the same issue with an ESP32 DEVKITV1. MD5 error and upload not working. 
Solved it with the command:  esptool write_flash_status --non-volatile 0
Need to run in the directory of esptool.exe
The port is autodetected, only connect one ESP! ;)

Thanks for the help!

alexlmhw, 2023-05-05:
https://docs.espressif.com/projects/esptool/en/latest/esp32/espefuse/set-flash-voltage-cmd.html
we had the same issue with an ESP32 WROVER-E.
Solved it with the command:  espefuse.py set_flash_voltage 3.3V

Not working for us: esptool --port write_flash_status --non-volatile 0

UsamaArshad16, 2023-07-14:
@alexlmhw It really worked
thank u so much 
espefuse.py --port /dev/ttyACM0 set_flash_voltage 3.3V
in my case

stsalyuk, 2023-07-14:
Was having the same issue and 

`espefuse.py set_flash_voltage 3.3V` 

worked for me as well

Pranitk1501, 2023-07-22:
None of the solutions are working for me. Erase flash takes around 13.5s so can't say if hardware is faulty. I am using ESP32 Wrover Devkit  and uploading code directly using USB cable.
Flash Status value is - 0x0000
I have already burned flash voltage to 3.3v

VeneKone, 2023-11-02:
> > > There is a way to reset all flash content? like factory defaults?
> > 
> > 
> > Yes, using `esptool.py erase_flash`
> 
> how to use this command?

1) First ensure that you've installed `esptool` by using the command `pip install esptool` in the command prompt. Check the exact syntax for using `pip` based on your operating system. If `esptool` is already installed, there's no harm in running the command again, as the command prompt will show something like "requirement already satisfied at..."

2) Once you're sure that `esptool` is installed, *then* run the `esptool --port COM# write_flash_status --non-volatile 0` command (replace `#` with the appropriate serial port number as shown by your Arduino IDE) in the command prompt.

3) One possible issue is: "Port does not exist" error. This happens if some other software is using that Serial COM port. Most of the times it's just the Arduino IDE, so close that window (and any other open Arduino IDE windows) before running the command.

NOTE: I'm posting this for the benefit of anyone who goes through this conversation in the future. I'm assuming that you've already figured out how to use the command, given that your message is more than 3-4 months old.

EthanC823, 2023-12-21:
> 1. First ensure that you've installed `esptool` by using the command `pip install esptool` in the command prompt. Check the exact syntax for using `pip` based on your operating system. If `esptool` [...]

I'm very new to all of this so I don't know if this works in VSCode esp-idf but I tried it anyway. I get the following error when running esptool --port COM3 write_flash_status --non-volatile 0 in the VSCode terminal.

[image]

I've used the command pip install esptool and it installs properly. When running the command again I get "Requirement already satisfied:" at the start of each line.

The problem is resolved when I disconnect everything from the VIN pin, but I'm still  wondering if the problem can be resolved in the terminal?

infiniteglory, 2024-05-05:
> 奥拉·卢卡斯！
> 
> 我通常会因为闪烁不良而发生这种情况，例如质量差或 USB 电缆太长，或者当有更多东西共享电源时......另一个提示，尝试使用波特率 115200 而不是 921600

thank you answer  The error occurs because I'm trying sharing power with oled when I flash the code

melyux, 2025-02-23:
Changed cables and it fixed this.