espressif/arduino-esp32 #1891: WiFiClientEnterprise.ino does not work for some Eduroam networks

zhxq, 2018-09-24:
### Hardware:
Board:							Node32s
Core Installation/update date:			On-Board version
IDE name:							Arduino 1.8.7
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						921600
Computer OS:                                            Windows 10

### Description:
Eduroam WPA-2 Enterprise is not working using the offical Eduroam code. Currently in New York University. Seems these codes work fine for some other universities as stated in @martinius96's own repo (the code contributor of this part of code). It does not work on either eduroam provided by NYU nor private network of NYU (also WPA2-Enterprise). I believe they share the same Wi-Fi configuration. I'm sure that my credentials are correct, because I tried to use the exact same credentials to connect to these networks again so that I can make sure the credentials in the code is correct.

As far as I know, "nyu-legacy" (2.4G) and "eduroam" provided by NYU is using WPA2-Enterprise and PEAP+MS-CHAPv2.

Connecting to normal WPA2-PSK network has no problem at all.

BTW, I do see some people can provide debug messages like 

> [D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
> Code: 0
> .E (3569) wpa: Method private structure allocated failure
> .[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
> [W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 2 - AUTH_EXPIRE
> .E (7118) wpa: Method private structure allocated failure
> [D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
> [W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 2 - AUTH_EXPIRE
> ..E (10691) wpa: Method private structure allocated failure

in issue #1381. How to enable this kind of debug message in Arduino IDE as well? Thanks!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
Filenames has been given under Debug Messages. Nothing changed excepted added `esp_log_level_set("*", ESP_LOG_VERBOSE);` for showing the debug info and changed Wi-Fi credentials (username/password).

### Debug Messages:

####Using the WPA2-Enterprise Eduroam in esp32 (offical) repo (https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEnterprise/WiFiClientEnterprise.ino):

Connecting to network: eduroam
I (29) wifi: wifi driver task: 3ffb7240, prio:23, stack:4096, core=0
I (99) wifi: wifi firmware version: 4fd6a2c
I (99) wifi: config NVS flash: enabled
I (102) wifi: config nano formating: disabled
I (126) wifi: Init dynamic tx buffer num: 32
I (126) wifi: Init data frame dynamic rx buffer num: 10
I (126) wifi: Init management frame dynamic rx buffer num: 10
I (129) wifi: Init static rx buffer size: 1600
I (133) wifi: Init static rx buffer num: 4
I (137) wifi: Init dynamic rx buffer num: 10
I (200) wifi: mode : sta (80:7d:3a:c5:d6:04)
I (201) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

I (332) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
.I (1063) wifi: state: init -> auth (b0)
I (1065) wifi: state: auth -> assoc (0)
..I (2065) wifi: state: assoc -> init (4)
I (2065) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
...................(and dots afterwards)

####Using the WPA2-Enterprise Eduroam in @martinius96's repo (https://github.com/martinius96/ESP32-Eduroam):

Connecting to: 
eduroam
I (181) wpa: WPA2 ENTERPRISE VERSION: [v2.0] enable

Mode: STA
Channel: 1
SSID (7): eduroam
Passphrase (0): 
BSSID set: 0
I (350) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
.I (1081) wifi: state: init -> auth (b0)
I (1082) wifi: state: auth -> assoc (0)
I (1086) wifi: state: assoc -> run (10)
I (1086) wpa: wpa2_task prio:2, stack:6656

...I (2733) wpa: >>>>>wpa2 FIALED

I (2744) wpa: wpa2 task delete

........................................................I (31086) wifi: state: run -> init (2)
I (31087) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (31088) wifi: flush txq
I (31088) wifi: stop sw txq
I (31090) wifi: lmac stop hw txq
I (31098) wifi: mode : sta (80:7d:3a:c5:d6:04)
I (31220) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (31220) wifi: state: init -> auth (b0)
I (31222) wifi: state: auth -> assoc (0)
I (31225) wifi: state: assoc -> run (10)
I (31225) wpa: wpa2_task prio:2, stack:6656

....I (32862) wpa: >>>>>wpa2 FIALED

I (32872) wpa: wpa2 task delete

........................................................I (61229) wifi: state: run -> init (2)
I (61230) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (61231) wifi: flush txq
I (61231) wifi: stop sw txq
I (61233) wifi: lmac stop hw txq
I (61241) wifi: mode : sta (80:7d:3a:c5:d6:04)
.I (61363) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (61363) wifi: state: init -> auth (b0)
I (61365) wifi: state: auth -> assoc (0)
I (61374) wifi: state: assoc -> run (10)
I (61375) wpa: wpa2_task prio:2, stack:6656

...I (63014) wpa: >>>>>wpa2 FIALED

I (63024) wpa: wpa2 task delete

........................................................I (91375) wifi: state: run -> init (2)
I (91376) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (91377) wifi: flush txq
I (91377) wifi: stop sw txq
I (91379) wifi: lmac stop hw txq
I (91387) wifi: mode : sta (80:7d:3a:c5:d6:04)
I (91509) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (91509) wifi: state: init -> auth (b0)
.I (91511) wifi: state: auth -> assoc (0)
..I (92511) wifi: state: assoc -> init (4)
I (92512) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1

It seems that these two versions have different error reports as well - the official version will report no info after that one "bunch" (and just print dots), and @martinius96's version will give out "WPA2 FIALED"(offical typo LOL) and try to reconnect and then fail again and reconnect and so on.

martinius96, 2018-09-24:
Hey there. For additional info to serial monitor you must enable Info or Debug option in: Tools --> Debug Level (default is none). It will print info about connection to your serial monitor. Try it and then please send us screenshot.

zhxq, 2018-09-24:
@martinius96 I did find that in esp8266, but this option does not exist for esp32 boards in Arduino. I just simply don’t see the debug level option under tools. As I said, when using esp8266, that option exists under Tools menu.

martinius96, 2018-09-24:
@zhxq 
You were true. In ArduinoIDE if I choose in ArduinoIDE Node32s board, it really does not have Core Debug Level. But i was successful to compile code under it and upload it to my DOIT Devkit V1 board without any problem and connection was sucessful too. 
I was testing today again :)
Core Debug Level: Debug
[image]
I haven't found any debug "command" that you can use, try to change board and upload sketch with that option of debug level messages.

zhxq, 2018-09-24:
@martinius96 Just wonder which IDE you are using? So I can also change the debug level in there as well. I really hope to see those debug messages - they might give me a clue. BTW you said your code works, so which version of your code you are currently using? 2017/2018/2018-new in your repo or this official one? I think I will also buy some of your boards as well to control variable so I can know either the board is not working or the code is not working.

martinius96, 2018-09-24:
1. Working for users that had problem with sketch 2. Worked under EAP-TTLS network too , it got some error checks there during authentization and some settings:
https://github.com/martinius96/ESP32-Eduroam/blob/master/connect_eduroam_2018_new_fix.ino

2. working for me like 1. sketch, for almost all users too, I am using this one for months:
https://github.com/martinius96/ESP32-Eduroam/blob/master/connect_eduroam_2018.ino
This sketch is same as in official esp32 repo: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiClientEnterprise/WiFiClientEnterprise.ino

That ESP32 board that i own, it is the cheapest that you can buy. Connection to Eduroam worked on first time for me.
With 1. and 2. sketch I had same results in debug messages and in connection. Both worked for me.

For instance, as experiment I have tried to build own WPA2 Enterprise network at home with Zeroshell Linux and it weren't working with sketches. It failed on handshake and it was also PEAP+ MsCHAPv2 option. Worked only on another devies, PC, laptop, smartphone.
Result:
[image]
My ArduinoIDE version: 1.8.5 
Look there: https://github.com/espressif/arduino-esp32/issues/1381#issuecomment-412073401
This user was able to login with username only without university in identity. I have tried it now and it worked too.

zhxq, 2018-09-24:
@martinius96 I wonder which IDE you are using - so I can also change the debug level to “debug”. Thanks a lot for your detailed and timely reply!

martinius96, 2018-09-24:
There isn't option for debug to Arduino boards, only something for serial.print function that will give you some informations
. 
I am using that debug core level option only at ESP32 boards, where it is implemented right to Tools card. If you will choose my board: DOIT ESP32 DEVKIT V1 you will have under Tools that option for Core Level Debug, then only change it, upload sketch and watch serial monitor and it will print you these informations and also problem during authentization. 

---------------
I have tested sketch on 2 networks: Eduroam and local WPA2 Enterprise at school hostel. Both were working and both were under Linux Debian with Freeradius extension. Maybe there is some problem in network settings of your university and not in board or sketch.

zhxq, 2018-09-25:
@martinius96 I just tried to connect to the NYU network, and unfortunately, I found the failure reason is exactly "HANDSHAKE_TIMEOUT", the same are your failure reason. I think I might need to find an alternative way to connect to internet - I don't think I'll close the issue, since I believe there must be some other people having the same problem. 

I will continue trying to find ways to connect to this tricky network, and if I have any progress, I will definitely reply here again. I'm trying to request a root certificate of their Wi-Fi network, and I'm currently waiting for the response from NYU IT Desk. As I said, I'll update if there is any new progress.

martinius96, 2018-09-27:
@zhxq hello, today I was testing ESP32 Dev module for first time. I had problems with connect to Eduroam network. I used Debug Core Level for board. At first upload of eduroam sketch, there were error EXPIRE: ASSOC - I have never seen that error yet.
After next upload to board, there were "error": wifi: Set status to INIT
After some resets with EN button, it connected sucessfuly. 
@me-no-dev any suggestion for that INIT info, to set it correctly? 
[image]
P.S. I must say, that cheapest ESP32 board - Devkit v1 worked much better... And it have some cpu WROOM-32 as this one.

piyord, 2018-10-01:
Hi there,

I have similar problem with eduroam. 

I posted before similar issue:
https://github.com/espressif/arduino-esp32/issues/1686

With  `arduino-esp32` API version with date 24-06-2018 eduroam works great.
Later this API date, with 'IDF Update to aaf1239' eduroam fails to connect.
Maybe yo can test with this oldest API to test if your eduroam network connects succesful.

piyord, 2018-10-01:
I think the problem is about some incompatibility with wpa2 library on'esp-idf' framework and the `arduino-esp32` API

zhxq, 2018-10-01:
> Hi there,
>
> I have similar problem with eduroam.
>
> I posted before similar issue:
> #1686
>
> With `arduino-esp32` API version with date 24-06-2018 eduroam works great.
> Later this API date, with 'IDF [...]

I will try to install the older 2018-06-24 version API and try it tomorrow. If possible, can you please give me the commit #? Thanks!

martinius96, 2018-10-02:
@piyord hello there, you are true. Today I have updated my esp32 software via get application that is located in: C:/...Documents/Arduino/hardware/esp32/tools
After update i have on both boards errors: wifi: Set status to INIT and I am unable to connect to Eduroam network. 

A few days ago with my software from June I was able to connect to Eduroam network with DevKit v1 board without any problem and I had that wifi: Set status to INIT error on second board - Dev module, but it was able to connect on fifth attempt.
[image]
@me-no-dev will fix it i think, maybe he will revert it.

lbernstone, 2018-10-15:
get.exe does not update the arduino code.  It updates the xtensa-gcc framework and a couple other subcomponents.  You need to `git pull` to get the latest code from the repository.

martinius96, 2018-10-15:
Oh, understand, then I am still on June code.. But i don't know why last time i started that get.exe file i was unable to login more to Eduroam.. 
Thanks for explain.

closed 2019-08-15 by stale[bot]