espressif/arduino-esp32 #1894: 3rd party library tries to access SPI 'SPDR' and 'SPSR' registers --> not declared in this scope////

TLS1000, 2018-09-24:
Hi, I have a 3rd party library that works on Arduino mega, but fails on ESP32 core.
native Arduino SPI.h lib works, the ESP32version doesn't. 

SPDR and SPSR registers don't seem to be defined in ESP32 lib.
how and where do I need to define these registers?

code
`#include <Arduino.h>
#include <stdint.h>
#include "Linduino.h"
#include "LT_SPI.h"
#include "Userinterface.h"
#include "LTC68041.h"
#include <SPI.h>

static const int spiClk = 1000000; // 1 MHz
#define MOSI 18
#define MISO 21
#define SCK 19
#define LTC6804_CS 5      //SS

void setup()                  
{
  Serial.begin(19200);
  SPI.begin();
  LTC6804_initialize();  //Initialize LTC6804 hardware
  init_cfg();        //initialize the 6804 configuration array to be written
  print_menu();              
}

`

stickbreaker, 2018-09-25:
Rewrite Library. SPDR &SPSR are  AVR hardware registers. You have to use the SPI library on ESP32, Different hardware.

Chuck.

closed 2018-09-26 by TLS1000

TLS1000, 2018-09-27:
ok thanks, I've rewritten the library, mainly replacing the SPDR and SPSR statements with basic SPI.write() and SPI.transfer() statements; it compiles but I haven't tested yet. 

I'm getting this single error which I don't see a problem in my code:

> U:\NewPrimary\My Documents\Arduino\libraries\LTC68041_ESP\LTC68041_ESP.cpp: In function 'int8_t LTC6804_rdaux(uint8_t, uint8_t, uint16_t (*)[6])':
> 
> U:\NewPrimary\My Documents\Arduino\libraries\LTC68041_ESP\LTC68041_ESP.cpp:494:3: error: expected primary-expression before 'const'
> 
>    const uint8_t GPIO_IN_REG = 3;
> 
>    ^
> 
> exit status 1
> Error compiling for board DOIT ESP32 DEVKIT V1.

the function in the library:
```
int8_t LTC6804_rdaux(uint8_t reg, //Determines which GPIO voltage register is read back. 
					 uint8_t total_ic,//the number of ICs in the system
					 uint16_t aux_codes[][6]//A two dimensional array of the gpio voltage codes.
					 )
{

  const uint8_t NUM_RX_BYT = 8;
  const uint8_t BYT_IN_REG = 6;
  const uint8_t GPIO_IN_REG = 3;
  
  uint8_t *data;
  uint8_t data_counter = 0; 
  int8_t pec_error = 0;
  uint16_t parsed_aux;
  uint16_t received_pec;
  uint16_t data_pec;
  data = (uint8_t *) malloc((NUM_RX_BYT*total_ic)*sizeof(uint8_t));
  //1.a
  if (reg == 0)
  {
	//a.i
    for(uint8_t gpio_reg = 1; gpio_reg<3; gpio_reg++)		 	   		 			//executes once for each of the LTC6804 aux voltage registers
    {
      data_counter = 0;
      LTC6804_rdaux_reg(gpio_reg, total_ic,data);									//Reads the raw auxiliary register data into the data[] array
	  
      for (uint8_t current_ic = 0 ; current_ic < total_ic; current_ic++) 			// executes for every LTC6804 in the daisy chain
      {																 	  			// current_ic is used as the IC counter
	  
        //a.ii
		for(uint8_t current_gpio = 0; current_gpio< GPIO_IN_REG; current_gpio++)	// This loop parses the read back data into GPIO voltages, it 
        {														   		  			// loops once for each of the 3 gpio voltage codes in the register 
          
		  parsed_aux = data[data_counter] + (data[data_counter+1]<<8);              //Each gpio codes is received as two bytes and is combined to
																				    // create the parsed gpio voltage code
																					
          aux_codes[current_ic][current_gpio +((gpio_reg-1)*GPIO_IN_REG)] = parsed_aux;
          data_counter=data_counter+2;												//Because gpio voltage codes are two bytes the data counter
																					//must increment by two for each parsed gpio voltage code
		  
        }
		//a.iii
        received_pec = (data[data_counter]<<8)+ data[data_counter+1]; 				 //The received PEC for the current_ic is transmitted as the 7th and 8th
																				     //after the 6 gpio voltage data bytes
        data_pec = pec15_calc(BYT_IN_REG, &data[current_ic*NUM_RX_BYT]);
        if(received_pec != data_pec)
        {
          pec_error = -1;															//The pec_error variable is simply set negative if any PEC errors 
																					//are detected in the received serial data
        }
       
        data_counter=data_counter+2;												//Because the transmitted PEC code is 2 bytes long the data_counter
																					//must be incremented by 2 bytes to point to the next ICs gpio voltage data
      }
   

    }
  
  }
  else
  {
	//b.i
    LTC6804_rdaux_reg(reg, total_ic, data);
    for (int current_ic = 0 ; current_ic < total_ic; current_ic++) 			  		// executes for every LTC6804 in the daisy chain
    {							   									          		// current_ic is used as an IC counter
	
		//b.ii
		for(int current_gpio = 0; current_gpio<GPIO_IN_REG; current_gpio++)  	 	// This loop parses the read back data. Loops 
		{						 											  		// once for each aux voltage in the register 
		
			parsed_aux = (data[data_counter] + (data[data_counter+1]<<8));    		//Each gpio codes is received as two bytes and is combined to
																					// create the parsed gpio voltage code
			aux_codes[current_ic][current_gpio +((reg-1)*GPIO_IN_REG)] = parsed_aux;
			data_counter=data_counter+2;									 		//Because gpio voltage codes are two bytes the data counter
																					//must increment by two for each parsed gpio voltage code
		}
		//b.iii
		received_pec = (data[data_counter]<<8) + data[data_counter+1]; 				 //The received PEC for the current_ic is transmitted as the 7th and 8th
																				     //after the 6 gpio voltage data bytes
        data_pec = pec15_calc(BYT_IN_REG, &data[current_ic*NUM_RX_BYT]);
        if(received_pec != data_pec)
        {
          pec_error = -1;													   		//The pec_error variable is simply set negative if any PEC errors 
																					//are detected in the received serial data
        }
        
		data_counter=data_counter+2;												//Because the transmitted PEC code is 2 bytes long the data_counter
																					//must be incremented by 2 bytes to point to the next ICs gpio voltage data
	}
  }
  free(data);
  return (pec_error);
}
```