espressif/arduino-esp32 #1895: Tone function not working in Timer interrupt

Autobot86, 2018-09-24:
I try to simple code code for pizzo buzzer working in esp32 timer interrupt but this code not working please help me

#include<Tone32.h>
boolean Timer_Flag = false;
const int Buzzer = 25;

hw_timer_t * timer1 = NULL;

void IRAM_ATTR onTimer1(void) {
Timer_Flag = !Timer_Flag;
if(Timer_Flag)
tone(Buzzer, 5000, 800,0);
else
noTone(Buzzer);

}

void user_init(void) {
/* Use 1st timer of 4 /
/ 1 tick take 1/(80MHZ/80) = 1us so we set divider 80 and count up */

timer1 = timerBegin(0, 80, true);
/* Attach onTimer function to our timer */

timerAttachInterrupt(timer1, &onTimer1, true);
/* Set alarm to call onTimer function every second 1 tick is 1us
=> 1 second is 1000000us /
/ Repeat the alarm (third parameter) */

timerAlarmWrite(timer1, 100000 , true);

timerAlarmEnable(timer1);
}
void setup() {
// put your setup code here, to run once:
Serial.begin(115200);
pinMode(Buzzer, OUTPUT);
user_init();

}
void loop() {

}

lbernstone, 2018-09-24:
Don't do that.  Set a flag in the ISR and process it in the main loop.

Autobot86, 2018-09-25:
But i worked this same logic on esp8266 and gives result by using OS Timer interrupt..??? then esp32 not responding this logic

Autobot86, 2018-09-26:
Am using 2 function in main loop and that execution time tooks approximately 1sec for single loop so it affects buzzer tone how shall i made possible also i used multitasking concept but fails that too?????

lbernstone, 2018-09-30:
#include <Ticker.h>
#include <Tone32.h>

Ticker tkFlip;
bool setTone, toneOn;
const int Buzzer = 25;

void switchTone() {
  setTone = true;
}

void setup() {
  tkFlip.attach(10,switchTone);
  setTone = true;
}

void loop() {
  if (setTone){
    if (toneOn) {
      noTone(Buzzer);
    } else {
      tone(Buzzer, NOTE_C4);
    }
    setTone = false;
    toneOn = !toneOn;
  }
}

closed 2019-08-15 by stale[bot]