espressif/arduino-esp32 #1898: Managment Frame Protection (802.11w) is supported?

Stasikiii, 2018-09-24:
Subj.

lbernstone, 2018-09-24:
https://en.wikipedia.org/wiki/LwIP

Stasikiii, 2018-09-25:
@lbernstone 
You can treat your answer as you please.
I'll try to reconfigure the question
If you select the option as shown here
https://www.cisco.com/c/dam/en/us/td/i/200001-300000/220001-230000/225001-226000/225274.tif/_jcr_content/renditions/225274.jpg
then can esp32 on arduino get connection to this AP?

Stasikiii, 2018-10-03:
So noone knows? 
This is sad cuse devices without 802.11w support (Managment Frame Protection) can be blocked with 2$ jammer. :(
if you don't trust me just try it for 4 steps and any wifi device without 802.11 support will be blocked
http://deauth.me/

Maby it is posseble to change some functions in the libs for blocks deauth requests?
In linux (routers side) for example progs can blocks deauth on core level for some devices with full mac layer support like broadcome or realtek. Bit client must ignore deauth too.

Any thoughts on this?
Cheers.

closed 2019-08-15 by stale[bot]

kwinz, 2021-01-28:
It is now, see https://github.com/espressif/esp-idf/releases/tag/v4.0.1