espressif/arduino-esp32 #1899: Unable to make the board work on OS X and Raspberry PI

riteshapatel, 2018-09-25:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do. - `Bought ESP32-WROOM from Amazon. Followed install instructions from Github README for OS X`. 
Describe your system( Hardware, computer, O/S, core version, environment) - `OS X and Raspberry PI 3 B+`
Describe what is failing - `Installed ESP32 by following GitHub README. Tried with Node32s as well as NodeMCU-32s boards. Both are failing when trying to compile and upload a bare bone example. On Mac Port is showing up as `/dev/cu.Bluetooth-Incoming-Port` on `Raspberry Pi 3` the port is showing up as `/dev/tty...`. `
Show the shortest possible code that will duplicate the error: `Bluetooth does not work.  Keeps showing Timed out waiting for packet header. `
Show the EXACT error message(it doesn't work is not enough)

```
/Users/ritesh/Documents/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c: In function 'uartResizeRxBuffer':
/Users/ritesh/Documents/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c:245:9: warning: 'return' with no value, in function returning non-void
         return;
         ^
Sketch uses 191764 bytes (14%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13332 bytes (4%) of dynamic memory, leaving 314348 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.5.0
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

```
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

riteshapatel, 2018-09-25:
I did push `boot + en`, not sure exactly what you mean by `try flash`. Newbie...

stickbreaker, 2018-09-25:
@riteshapatel simplest answer is that the port you selected is invalid.

Are you connecting directly to the board? (wired USB cable?)

If I interpret this correctly:
>Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

It says that you are trying to connect over bluetooth?

This would only work if you have a bluetooth to usb serial converter? (I don't know of any that exist.)

Why don't you plug a USB cable directly into the board?

What version of the Arduino environment are you using?

Chuck.

riteshapatel, 2018-09-25:
I am using `Arduino 1.8.7`

stickbreaker, 2018-09-25:
@riteshapatel when you connect the usb cable, does the list of available Serial sources change?

I agree that you probably have a driver problem. 

You might want to try the solution in #1376 

Chuck.

riteshapatel, 2018-09-25:
Ok a step further. Replaced the cable and I see two ports now.

[image]

And also did the hard reset by pressing both switches / releasing EN and then Boot.

I see the messages below.

```
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 3c:71:bf:84:cd:8c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 7230.4 kbit/s)...
Hash of data verified.
Compressed 15664 bytes to 10060...

Writing at 0x00001000... (100 %)
Wrote 15664 bytes (10060 compressed) at 0x00001000 in 0.9 seconds (effective 140.2 kbit/s)...
Hash of data verified.
Compressed 196448 bytes to 97932...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 196448 bytes (97932 compressed) at 0x00010000 in 8.7 seconds (effective 181.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1381.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

And trying to upload this sample program.

```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Hello from DFRobot ESP-WROOM-32");
  delay(1000);
}
```

Still can't connect.

riteshapatel, 2018-09-25:
Ok, so I can see the messages in `serial monitor` but only after `hard resetting`. To upload another program say for example `Blink` I have to press `boot` again and then only it blinks the lights. It'd be annoying to keep pressing `boot` each time I want to upload a new program. Is there an easier fix? 

Thx

stickbreaker, 2018-09-25:
The boot switch duplicates a circuit connected to the DTR amd RTS signals from the USB interface. Either your board is not built correctly, or the  DTR/RTS functions of your Mac device driver are not correctly setup.

I would recommend you look through your driver configuration, specifically for anything relating to DTR and RTS.  I have not experience with Mac's so that's as much as I know.

Chuck.

ozarchie, 2018-11-12:
The message above is a compile warning:
`/Users/John/Documents/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c: In function 'uartResizeRxBuffer':
/Users/John/Documents/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-uart.c:245:9: warning: 'return' with no value, in function returning non-void
         return;
`
It is from esp32-hal-uart.c, line 249:
`size_t uartResizeRxBuffer(uart_t * uart, size_t new_size) {
    if(uart == NULL) {
        return;
    }
`
The function is defined as returning 'size_t' but you can see it just returns!

closed 2019-08-15 by stale[bot]