espressif/arduino-esp32 #1900: esp_sleep_enable_gpio_wakeup() is missing

rayvt, 2018-09-26:
According to api-reference/system/sleep_modes.rst, I should be able to do GPIO wakeup (light sleep only).

Using the examples, my code is:
#include <esp_sleep.h>
gpio_wakeup_enable(GPIO_NUM_12, GPIO_INTR_LOW_LEVEL);
esp_sleep_enable_gpio_wakeup();
esp_light_sleep_start();

But esp_sleep_enable_gpio_wakeup() is not defined.  When I compile it, the error message is:
error: 'esp_sleep_enable_gpio_wakeup' was not declared in this scope
   esp_sleep_enable_gpio_wakeup();

lbernstone, 2018-09-26:
I'm unable to reproduce (aka works for me).  Can you turn on verbose compiling in your preferences and make sure it is pulling the correct esp_sleep.h?

rayvt, 2018-09-26:
That was it.  The esp_sleep.h I have in .\packages\esp32\hardware\esp32\1.0.0\tools\sdk\include\esp32 is dated 7/28/2018 and has 325 lines.  The one at https://raw.githubusercontent.com/espressif/arduino-esp32/master/tools/sdk/include/esp32/esp_sleep.h has 371 lines, and has the lines for esp_sleep_enable_gpio_wakeup().  

In Arduino IDE Boards Manager esp32 version shows 1.0.0.  I removed it and re-installed it and it still has the 7/28/2018 version of esp_sleep.h.   When I manually replaced it with the one from github it compiled okay.   Looks like something hasn't been updated so the Boards Manager doesn't pick up the latest arduino-esp32

lbernstone, 2018-09-26:
boards manager only picks up releases.  The repo changes daily.  The release version did not include light sleep yet.

rayvt, 2018-09-30:
Works on arduino-esp-master dated 9/20/2018 11:40 PM

closed 2018-09-30 by rayvt