espressif/arduino-esp32 #1901: digitalWrite vs REG_WRITE on pin13 - why is it different?

OhSoGood, 2018-09-26:
Hi,

Can anybody explain me why digitalWrite and REG_WRITE have different behavior on some pin?

On my devkit1, if I do "pinMode(13, OUTPUT); digitalWrite(13, HIGH);", I observe 3.3V on pin13.
But if I do "REG_WRITE(GPIO_ENABLE_REG, BIT13); REG_WRITE(GPIO_OUT_REG, BIT13);" (or REG_WRITE(GPIO_OUT_W1TS_REG, BIT13) ), pin13 stays at 0V

On the contrary, if I try the same on BIT27, pin13 gets 3.3V with REG_WRITE.

What is the reason for this? I looked at the board pin layout, but got no concrete clue.

Thank you!

francescolavra, 2018-10-27:
The reason is that some pins are not assigned to the GPIO function by default, so if you want to use a pin as a GPIO you have to make sure first that the pin multiplexer is configured correctly for that pin to function as a GPIO. That's what the `pinMode()` function does. If you call `REG_WRITE(GPIO_OUT_REG, BIT13)` after `pinMode(13, OUTPUT)`, you will see that the pin is correctly set at 3.3V.

OhSoGood, 2018-10-27:
Oh... You mean that the initialization I thought I was doing with `REG_WRITE(GPIO_ENABLE_REG, BIT13)` is not equivalent to `pinMode(13, OUTPUT)` ?
If so, what is the purpose of `REG_WRITE(GPIO_ENABLE_REG, xx=)` and should I do it in addition to `pinMode(13, OUTPUT)` ?
Thank you

francescolavra, 2018-10-27:
`pinMode(13, OUTPUT)` does other stuff in addition to setting bit 13 in `GPIO_ENABLE_REG`, and is not equivalent to `REG_WRITE(GPIO_ENABLE_REG, BIT13)`.
The purpose of `REG_WRITE(GPIO_ENABLE_REG, xx=)` is to configure GPIOs as inputs or outputs; you don't need to call it after pinMode(), because pinMode() already does all that's needed to configure a GPIO as output.

closed 2018-10-27 by OhSoGood