espressif/arduino-esp32 #1902: SD-card Append Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

vanvince, 2018-09-26:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE/Visual Studio Code 
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					961200
Computer OS:                                          Windows 10

### Description:
Hello guys,
i'm trying to save onto a text file some data about my control unit. There are several realtime task (sensors setup task and control task). In particular my project works as shown below:
- Setup of 9-axis abs orientation (BNO055);
- Setup of GSM/GPS module (Ai-Thinker A7);
- Setup of LOG file (create and write the header onto file) 
When setup is completed the Control part starts:
- Control: BNO055 control task starts its job and only when there is a dangerous angle it create the GSM/GPS control task. Then it suspends itself.
- Contol: GSM/GPS takes gps data and then send a sms to a certain phone number. After that resume the BNO055 control task.
Every task is pinned to a particular core. 
Just like this the entire project is working good, but when I put an append task to the log file everything crash with **Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.**

As you can see in the exception decode section the problem is due to a wrong print. I tried to write onto the log file using myFile.write despite of myFile.print but **write accepts only uint8_t data type** and this is not good for my intentions. I have to write double data type or char data type, so myfile.write isn't good at all. I tried also to execute the append inside an another control task but this not working at all. 

The realtime task code for the sdcard log file is shown below:

### Sketch:  
```cpp

// Setup task for datalog
static void datalog_setuptask(void* parameter){
  
  Serial.println("---Setup DATALOG Task--- ");
  Serial.println("\t[DATALOGSETUPTASK]Initializing SD card");
  Serial.print("\t[DATALOGSETUPTASK]DATALOG Task is running on core:");Serial.println(xPortGetCoreID());
  vTaskDelay(100/portTICK_PERIOD_MS);

  if(!SDInit(CHIPSELECT)){
    Serial.println("\t[DATALOGSETUPTASK]Init SD failed");
    vTaskDelay(100/portTICK_PERIOD_MS);
    vTaskDelete(DLogSetupTask);
  }
  else { 
    Serial.println("\t[DATALOGSETUPTASK]Init SD succeed");

    if(SDcreateLog(path)){
      Serial.println("\t[DATALOGSETUPTASK]Init log file");
      vTaskDelay(100/portTICK_PERIOD_MS);
      
      if(SDwriteLog(path,header))
        Serial.println("\t[DATALOGSETUPTASK]write header succeed");
      else Serial.println("\t[DATALOGSETUPTASK]write failed");

      if(SDappendLog(path,subheader))
        Serial.println("\t[DATALOGSETUPTASK]append subheader succeed");
      else Serial.println("\t[DATALOGSETUPTASK]append failed");
    }
    else Serial.println("\t[DATALOGSETUPTASK]File already exists");

    vTaskDelay(100/portTICK_PERIOD_MS);
    vTaskDelete(DLogSetupTask);
  }
}

//Append data Task for datalog
static void datalog_appendtask(void* parameter){

  Serial.println("---Append Log DATALOG Task--- ");
  Serial.println("\t[DATALOGTASK]Starting append data to LOG file");
  vTaskDelay(100/portTICK_PERIOD_MS);

  if(SDappendLog(path,test))
    Serial.println("succeed");
  else Serial.println("failed, like always. LOSER");
  
  vTaskDelay(100/portTICK_PERIOD_MS);
  vTaskDelete(DLogAppendTask);
  
}

// file init,create,write and append 
bool SDInit(uint8_t chips){

  if(!SD.begin(chips))
    return false;
  else
    return true;

}

bool SDcreateLog(char* path){
    
  if (!SD.exists(path)) {
    // Create a new file 
    myFile=SD.open(path,FILE_APPEND);
    myFile.close();
    return true;
  }
  else 
      return false;        
}

bool SDwriteLog(char* path,char* wdata){

  myFile=SD.open(path,FILE_APPEND);
  if(myFile){
      myFile.println(wdata);
      myFile.close();
      return true;
  }
  else 
      return false;
}

bool SDappendLog(char* path,char* wdata){

  myFile=SD.open(path,FILE_APPEND);
  if(myFile){
      myFile.print("");
      myFile.println(wdata);
      myFile.close();
      return true;
  }
  else 
      return false;
}
```

### Debug Messages:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x400d5484  PS      : 0x00060430  A0      : 0x800d54a0  A1      : 0x3ffd33f0  
A2      : 0x3ffc2320  A3      : 0x3f401cd9  A4      : 0x00000000  A5      : 0x3f401779  
A6      : 0x3f4012a4  A7      : 0x3ffd2500  A8      : 0x00000000  A9      : 0x3ffd33e0  
A10     : 0x00000000  A11     : 0x3f401cd9  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400d5484:0x3ffd33f0 0x400d549d:0x3ffd3410 0x400d18ab:0x3ffd3430 0x400d198e:0x3ffd3470

```
### Exception Decoder
```
PC: 0x400d5484: Print::write(char const*) at C:\Users\Mariano\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.h line 66
EXCVADDR: 0x0000000c

Decoding stack results
0x400d5484: Print::write(char const*) at C:\Users\Mariano\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Print.h line 66
0x400d549d: Print::print(char const*) at C:\Users\Mariano\Documents\Arduino\hardware\espressif\esp32\cores\esp32\Print.cpp line 84
0x400d18ab: SDappendLog(char*, char*) at C:\Users\Mariano\Documents\Arduino\SecureTrac/SecureTrac.ino line 1118
0x400d198e: datalog_appendtask(void*) at C:\Users\Mariano\Documents\Arduino\SecureTrac/SecureTrac.ino line 498
```

beats0126, 2018-09-27:
i think u may try to put the function to core 0 loop there
because few month ago i meet the same problem.
after that i move the function to core 0 loop then dont have the problem.

before this the issue i know is the http request, wifi function, SD card cannot do in core 1
so i arrange my code from core 1 to core 0 and it solved.

i not sure the espressif solved the problem already or no, but using the latest stable version, it look like can do http request at core 1 already (this one i cannot confirm because that is my friend told me)

vanvince, 2018-10-02:
Hi,
if I put the datalog_appendtask in the core 0 I detect the very same issue. I really don't understand why. Everything else is working fine.
Below I put the realtime code for the whole tasks, maybe there is something wrong that I didn't saw:
**Setup ()**
```
xTaskCreatePinnedToCore(&bno055_setuptask,"bno055_setuptask",STACKSIZE_2k5,NULL,1,&BNO055SetupTask,0);
xTaskCreatePinnedToCore(&a7gpsgsm_setuptask,"a7gpsgsm_setuptask",configMINIMAL_STACK_SIZE,NULL,1,&GPSSetupTask,1);
xTaskCreatePinnedToCore(&datalog_setuptask,"datalog_setuptask",STACKSIZE_2k5,NULL,1,&DLogSetupTask,0);
```
**bno055 imu read task**
```
//reading imu data from bno055
... do something
if ((imu[X_AXIS]>THRESHOLD || imu[Y_AXIS]>THRESHOLD) || (imu[X_AXIS]<nTHRESHOLD || imu[Y_AXIS]<nTHRESHOLD)) {
      if(!isOVerturned){
      vTaskDelay(100/portTICK_PERIOD_MS);
      
      Serial.println("\t\t\t[BNO055Task]SecureTrac ->WARNING<-");
      Serial.print("\t\t\t[BNO055Task]ROLL:");Serial.print(imu[X_AXIS]);Serial.print(" PITCH:");Serial.print(imu[Y_AXIS]);Serial.print(" YAW:");Serial.println(imu[Z_AXIS]);
      digitalWrite(LED_WARNING,HIGH);
      isOVerturned=true;
      xTaskCreatePinnedToCore(&a7gpsgsm_task,"a7gpsgsm_task",STACKSIZE_1K,NULL,8,&GPSTask,1);
      xTaskCreatePinnedToCore(&datalog_appendtask,"datalog_appendtask",4096,NULL,4,&DLogAppendTask,1);
... do something
```

closed 2019-08-15 by stale[bot]