espressif/arduino-esp32 #1905: WiFiClientSecure.connected() does not return the correct value

r-downing, 2018-09-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                          ?no?
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
When making a request using HTTP/1.0, the socket should close after the request is done. However, client.connected() will always return true.

Looking at the code for [connected()](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L202), it seems like it trys to read 0 bytes, assuming that the read will fail and set the internal _connected variable to false. However, in the [read(*,*)](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp#L166) function it says if the read size is 0, return 0 before attempting the read. I've tried modifying WiFiClientSecure.cpp and putting the raw read in the connected function but that didn't seem to work.

This sketch should open the connection, print out the whole google home page down to the &lt;/html>, then close the connection and start the loop over again, but it gets stuck thinking the client is still connected. It works as expected with the regular WiFiClient.

### Sketch: 
```cpp

#include <ssl_client.h>
#include <WiFiClientSecure.h>
#include <WiFi.h>

#define USE_SECURE

#ifdef USE_SECURE
#define PORT 443
#else
#define PORT 80
#endif

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

#ifdef USE_SECURE
  WiFiClientSecure client;
#else
  WiFiClient client;
#endif
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.print("GET / HTTP/1.0\r\n\r\n");
  while (client.connected()) {
    if (client.available()) {
      Serial.write(client.read());
    }
  }
  //never gets here with WiFiClientSecure, even though we are reading all data out of the client in a while loop...
  Serial.println("\n\nclosing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.begin("SSID", "PASSWORD");
  Serial.print("connecting to wifi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

}

void loop()
{
  testClient("google.com", PORT);

  delay(10000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

r-downing, 2018-09-30:
My original goal was to make a UART to ethernet bridge, to be controlled by another device. I've found that when I send "GET / HTTP/1.0\r\n\r\n" the first time, it works. Then when I send that same string again, it will transmit the string, up to at least the first '\r' or '\n' before it detects the connection has closed and starts the loop again. It also prints out an invalid character, indicating that client.available() is giving the incorrect value as well.

```cpp

#include <ssl_client.h>
#include <WiFiClientSecure.h>
#include <WiFi.h>

#define SSID "ssid"
#define PASSWORD "password"

#define USE_SECURE

#ifdef USE_SECURE
#define PORT 443
#else
#define PORT 80
#endif

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

#ifdef USE_SECURE
  WiFiClientSecure client;
#else
  WiFiClient client;
#endif
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  //  client.print("GET / HTTP/1.0\r\n\r\n");
  while (client.connected()) {
    if (client.available()) {
      Serial.write(client.read());
    }
    if (Serial.available()) {
      char b = Serial.read();
      Serial.print(b);
      client.write(b);
    }
  }
  //never gets here with WiFiClientSecure, even though we are reading all data out of the client in a while loop...
  Serial.println("\n\nclosing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.begin(SSID, PASSWORD);
  Serial.print("connecting to wifi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

}

void loop()
{
  testClient("google.com", PORT);
}

```

nae9, 2018-10-08:
This problem appiared in new version of esp32 libs. Old version work fine. It seemed that code developers not test release code.

r-downing, 2018-10-08:
I see... Which version should I download if I want this to work? According to the Arduino Boards Manager I have 1.0.0

nae9, 2018-10-10:
Early i use installation in UserDocuments/Arduino/Hardware/espressif/esp32/...
and i have bakup of this folder [https://drive.google.com/file/d/1-4Ab6572dV7KypVEbPMz9a8yU7l_Db8c/view?usp=sharing](url). 

Copy this into UserDocuments/Arduino/hardware/espressif/esp32/
If you use additional libs then put it in UserDocuments/Arduino/libraries/
Your skethes put in UserDocuments/Arduino/
Set in arduino IDE props sketch folder as "UserDocuments/Arduino/" and reload IDE.

closed 2019-08-15 by stale[bot]