espressif/arduino-esp32 #1906: Partition Scheme definitions for esp32doit-devkit-v1 board

DusanLesan, 2018-09-28:
After reading about esp32 partitions and easy way to change it from Arduino IDE, I have found out that my board does not have Partition Scheme menu option. That seems to be defined in the boards.txt file and I have tried copying PartitionScheme lines from other board to esp32doit-devkit-v1:
```
esp32doit-devkit-v1.menu.PartitionScheme.default=Default
esp32doit-devkit-v1.menu.PartitionScheme.default.build.partitions=default
esp32doit-devkit-v1.menu.PartitionScheme.minimal=Minimal (2MB FLASH)
esp32doit-devkit-v1.menu.PartitionScheme.minimal.build.partitions=minimal
esp32doit-devkit-v1.menu.PartitionScheme.no_ota=No OTA (Large APP)
esp32doit-devkit-v1.menu.PartitionScheme.no_ota.build.partitions=no_ota
esp32doit-devkit-v1.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
esp32doit-devkit-v1.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (Large APPS with OTA)
esp32doit-devkit-v1.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
esp32doit-devkit-v1.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080
esp32doit-devkit-v1.menu.PartitionScheme.fatflash=16M Fat
esp32doit-devkit-v1.menu.PartitionScheme.fatflash.build.partitions=ffat
```
These lines enabled Partition Scheme menu in my Arduino IDE and I have tested it with default and no_ota schemes. I have not found any issues with it.

What is the process of implementing this change?

I have tested my partitions with this code:
```
#include <esp_partition.h>

void setup() {
  Serial.begin(115200);
  const esp_partition_t *partition = NULL;
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_APP_OTA_0, "app0");
  if (partition != NULL) {
    Serial.printf("APP partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
  
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_APP, ESP_PARTITION_SUBTYPE_APP_OTA_1, "app1");
  if (partition != NULL)
  {
    Serial.printf("OTA partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
  else
  {
    Serial.printf("No OTA partiton found !\n");
  }
  
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, (esp_partition_subtype_t)0x99, "eeprom");
  if (partition != NULL) {
    Serial.printf("EEprom partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
  
  partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_SPIFFS, "spiffs");
  if (partition != NULL) {
    Serial.printf("SPIFFS partiton addr: 0x%08x; size: %d; label: %s\n", partition->address, partition->size, partition->label);
  }
}

void loop() {}
```

DusanLesan, 2018-09-28:
@lbernstone I agree completely if some other board definition does the same job. However, since it does exist here, I think full support is ideal.

atanisoft, 2018-09-28:
it would be better if partitions and debug levels could be moved away from boards and instead be made global to all boards...

DusanLesan, 2018-09-28:
@atanisoft I don't know if all board have the same memory capacity. If so, you are right. But I also wouldn't mind the total removal of some boards if definitions are the same.

atanisoft, 2018-09-28:
> @atanisoft I don't know if all board have the same memory capacity.

Most are of similar sizes from what I have seen, same for pin mappings. But people tend to pick a board entry from the menu to match whatever their board is actually called. For those boards that are generic (most) it would be good to have a common base set of values (like the devkit ones) that can then be extended by the rest of the boards. I'm not sure if the boards.txt format supports that though.

closed 2019-08-15 by stale[bot]

Le-Ze, 2022-01-03:
Hello.
I have an ESP-Wroom-32D board.
I'm using Arduino ID 1.8.19
When I choose ESP Dev Module, it doesn't work.
I have to use DOIT ESP32 Devkit V1
DOIT ESP32 DEVKIT V1 has no partition schemes.
Why doesn't he?
How to make?