espressif/arduino-esp32 #1907: Cache disabled but cached memory region accessed

anandpanchal, 2018-09-28:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

atanisoft, 2018-09-28:
instead of volatile for each variable you are using in the ISRs try using DRAM_ATTR instead. volatile doesn't move it to IRAM/DRAM which is required for access in ISRs.

anandpanchal, 2018-10-01:
volatile unsigned long fall_time1;                                                 //  time stamping for first ir receiver
volatile unsigned long fall_time2;                                                 //  time stamping for second ir receiver
volatile bool fall_flag1 = false;                                                       //  flag for first ir receiver
volatile bool fall_flag2 = false;                                                       //  flag for second ir receiver

#define duration 1000                                                                 //  time elapsed

unsigned long interval_1 = 0;                                                     //  time interval variable for ir receiver first                                 
unsigned long interval_2 = 0;                                                     //  time interval variable for ir receiver second

unsigned long interval  = 0;

hw_timer_t *timer0 = NULL;                                    
portMUX_TYPE timerMux0 = portMUX_INITIALIZER_UNLOCKED;

volatile bool ledstat = 0;

void IRAM_ATTR onTimer0()                                        // **************************************esp32 timer interrupt isr *****************//
{
  portENTER_CRITICAL_ISR(&timerMux0);
  ledstat = 1-ledstat;
  digitalWrite(23, ledstat);                                               // on/off pin no 23
  portEXIT_CRITICAL_ISR(&timerMux0);
  
}

void fall_isr1()                                                                           //  interrupt service routine for first ir rreceiver receiver
{
   
   volatile unsigned long last_1 = millis();                                  //  code to ignore the interrupts if occurs within one second       
   if ((last_1 - interval_1) > duration)
   {
   fall_flag1 = true;
   fall_time1 = last_1;
   interval_1 = last_1;
   }
}

void fall_isr2()                                                                      //  interrupt service routine for second ir receiver
{
   volatile unsigned long last_2 = millis();                             //  code to ignore the interrupts if occurs within one second
   if((last_2 - interval_2) > duration)
   {
   fall_flag2 = true;
   fall_time2 = last_2;
   interval_2 = last_2;
}
}

void setup() 
{  
  Serial.begin(115200);                                                                //  initilized arduino serial communication port
  pinMode(5, OUTPUT);                                                              //  set pin 5 as output. relay is attached to this pin
  pinMode(23, OUTPUT);                                                           //*************************ir transmitter pin****************************//
  digitalWrite(23 , LOW); 
  Serial.println("Monitoring interrupts: ");         
  pinMode(21, INPUT_PULLUP);                                                 //   set pin 21 as input. ir receiver 2 is attached to this pin                    
  pinMode(19, INPUT_PULLUP);                                                  //  set pin 19 as input. ir receiver 1 is attached to this pin.
  attachInterrupt(digitalPinToInterrupt(21), fall_isr2, FALLING);      //  notice the interrupt of second ir receiver at falling edge and call the ISR
  attachInterrupt(digitalPinToInterrupt(19), fall_isr1, FALLING);      //   notice the interrupt of first ir receiver at falling edge and call the ISR

  timer0 = timerBegin(0, 80, true);                                        //******************esp32 timer0*****************************//   
  timerAttachInterrupt(timer0, &onTimer0, true);  // edge triggered
  timerAlarmWrite(timer0, 14, true);   // 14 * 1us, autoreload true.
  timerAlarmEnable(timer0);
}

void loop() 
{

  unsigned long currentMillis = millis();
   
   if (fall_flag1 && fall_flag2)                                                           //  if interrupts occurs at both the pins
   {
       fall_flag1 = false;
       fall_flag2 = false;
       
       if ((fall_time2 > fall_time1) || (fall_time2 == fall_time1))           // if second beam has broken later, it means someone has entered inside and turn on the lights.
       {
        digitalWrite(5, LOW);
        interval = millis();
        
       }
       if (fall_time1 > fall_time2)                                                       //  if first beam has broken later, it means someone has left and turn off the lights.
       {
        digitalWrite(5, HIGH);
        
                
       }
       
   }

anandpanchal, 2018-10-01:
void loop() 
{

  unsigned long currentMillis = millis();
   
   if (fall_flag1 && fall_flag2)                                                           //  if interrupts occurs at both the pins
   {
       fall_flag1 = false;
       fall_flag2 = false;
       
       if ((fall_time2 > fall_time1) )           // if second beam has broken later, it means someone has entered inside and turn on the lights.
       {
        digitalWrite(5, HIGH);
        interval = millis();
        
       }
       if (fall_time1 > fall_time2)                                                       //  if first beam has broken later, it means someone has left and turn off the lights.
       {
        digitalWrite(5, LOW);
        
                
       }
       
   }

closed 2019-08-15 by stale[bot]