espressif/arduino-esp32 #1910: Bogus WDT error

rayvt, 2018-09-30:
Hardware: ESP32 Dev Module
Core: arduino-esp-master dated 9/20/2018 11:40 PM
Arduino IDE

### Description:
Occasionally bogus WDT error.  Possibly related to #1754 & #1881

On one run, it happened 12 times in 17,700 seconds.  883 sleeps at 20 seconds each.

In this simple code (below) it happens only once in a while.  In my real program that
does a lot of other stuff, it sometimes happens once in a while, sometimes happens every time
it does a lora.send, and sometimes the entire board locks up solid.

================================
Example Output:

Going to sleep.   ++++++++++++++++++++++++++++++++++++++++  (ts: 8798)
GPIO level: 1
Woke up:  !!!!!!!!!!!!!!!!! Slept  20000 msec   GPIO level: 1  (ts :8821)
E (8852) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (8852) task_wdt:  - IDLE0 (CPU 0)
E (8852) task_wdt: Tasks currently running:
E (8852) task_wdt: CPU 0: wifi
E (8852) task_wdt: CPU 1: IDLE1
GPIO level: 1 after debounce delay.  (ts: 8871)   uptime: 1708 s

Going to sleep.   ++++++++++++++++++++++++++++++++++++++++  (ts: 53431)
GPIO level: 1
E (53439) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (53439) task_wdt:  - IDLE0 (CPU 0)
E (53439) task_wdt: Tasks currently running:
E (53439) task_wdt: CPU 0: wifi
E (53439) task_wdt: CPU 1: loopTask
Woke up:  !!!!!!!!!!!!!!!!! Slept  20000 msec   GPIO level: 1  (ts :53477)
GPIO level: 1 after debounce delay.  (ts: 53527)   uptime: 13953 s

================================
### Sketch:  
Code that shows the problem:
WiFi.mode is WIFI_AP_STA,   but wifi is not connected and nothing is going on.

```
#define DOOR_PIN GPIO_NUM_12 

void loop()
{
  unsigned long t1;

  esp_sleep_enable_timer_wakeup(20 * 1000000); //20 seconds -- temp, for testing
  Serial.println("");
  Serial.printf("Going to sleep.   ++++++++++++++++++++++++++++++++++++++++  (ts: %lu)\n", esp_log_timestamp());
  Serial.printf("GPIO level: %d\n", digitalRead(DOOR_PIN));
  Serial.flush();   delay(10);  Serial.flush();
  t1 = millis();
  yield();  /* Try to evade bogus WDT alarm. */
  esp_light_sleep_start();
  yield();

  Serial.printf("Woke up:  !!!!!!!!!!!!!!!!! Slept %6lu msec   GPIO level: %d  (ts :%lu)\n",  millis() - t1, digitalRead(DOOR_PIN), esp_log_timestamp());
  delay(50);  // debounce
  Serial.printf("GPIO level: %d after debounce delay.  (ts: %lu)   uptime: %d s\n", digitalRead(DOOR_PIN), esp_log_timestamp(), millis()/1000);
}
```

zhivko, 2019-07-01:
Me having same bogus wdt messag:
```
E (296979) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (296979) task_wdt:  -  (CPU 0)
E (296979) task_wdt: Tasks currently running:
E (296979) task_wdt: CPU 0: IDLE0
E (296979) task_wdt: CPU 1: pidTask
E (301979) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (301979) task_wdt:  -  (CPU 0)
E (301979) task_wdt: Tasks currently running:
E (301979) task_wdt: CPU 0: IDLE0
E (301979) task_wdt: CPU 1: pidTask
```

closed 2019-09-13 by stale[bot]

nitko12, 2021-02-16:
Any new info?

I am experiencing something similiar, also when using light sleep with wifi, if that helps?