espressif/arduino-esp32 #1912: ESP32 Gateway & EVB - ETH Disconnect

HarterHorst, 2018-09-30:
Hi,
i have a problem with Olimex ESP32 EBV and Gateway. Code works, Ethernet connections works but i get a disconnect immediately. 

### Hardware:
Board:							Olimex ESP32 Gateway & EVB
Core Installation/update date:			latest
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                          no
Upload Speed:						115200
Computer OS:                                            Windows 10

### Description:
disconnects
`ETH Started
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH Disconnected
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH Disconnected
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH Disconnected
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH Disconnected
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH MAC: 30:AE:A4:1B:C1:FF, IPv4: 192.168.0.90, FULL_DUPLEX, 100Mbps
ETH Disconnected
`

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  ETH.config(IPAddress(192, 168, 0, 90),IPAddress(192, 168, 0, 1),IPAddress(255, 255, 255, 0));
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

DanKoloff, 2018-10-01:
Try to add DNS. Also check the suggestions here: https://github.com/espressif/arduino-esp32/issues/1762#issuecomment-413913021

HarterHorst, 2018-10-01:
I am sure that a missing dns is not the reason for the disconnections, but I tried of course. Same message.

`/*
    This sketch shows the Ethernet event usage

*/

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      Serial.print("DNS:\t\t");
      Serial.println(ETH.dnsIP());
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
  ETH.config(IPAddress(192, 168, 0, 90), IPAddress(192, 168, 0, 1), IPAddress(255, 255, 255, 0), IPAddress(192, 168, 0, 2), IPAddress(192, 168, 0, 1));
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}`

DanKoloff, 2018-10-02:
What happens if you use the default example and DHCP instead of configuring it manually with ETH.config?

HarterHorst, 2018-10-02:
With default example you probably mean this: [https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino](url)

I put it on but no changes. see below. I already changed the cable and next I will try another switch. Maybe there is an issue with link auto detection, because it's a 1GBit/s switch. Gonna dig through the basement to get a 100MBit/s switch. 

I am not familiar with the librarie ETH.h uses or which library is responsible to detect connect/disconnect. Can the Wifi lib deliver more details? 

```ets Jun 2016 ⸮⸮22:57C⸮HH⸮⸮⸮0x1 (POWŪ⸮⸮%UMQ),boot:⸮⸮b (SPReMQ}1SH_BOOU⸮Hh⸮⸮⸮⸮⸮sip: 0,⸮A%]A⸮0⸮e
clkE⸮⸮⸮0x0b⸮}⸮⸮⸮⸮0x00,d_dr⸮⸮⸮⸮0,c.z⸮⸮⸮⸮0⸮0,hd_d.⸮⸮⸮⸮0,}⸮⸮⸮⸮0⸮⸮
mode'T⸮ clk⸮⸮div:LC!녑⸮0x3fff00Lblen:4⸮!⸮+⸮⸮0x&⸮⸮⸮⸮⸮len:132SH⸮녑⸮0x400780⸮blen:77⸮
ho 0⸮⸮⸮⸮⸮12⸮⸮⸮⸮⸮4
!⸮+⸮⸮0x⸮80400⸮⸮⸮⸮6448C⸮⸮⸮⸮⸮0x400806Yj
ETH Started
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
ETH Connected
ETH Disconnected
```

DanKoloff, 2018-10-02:
Hmm I used the exact same code as you linked. Also compared the ETH.h file and they are the same. However, ETH.h has other includes inside (like WiFi.h and inside WiFi.h there are 12 more includes) and maybe some of these are outdated at your side. Have you updated the ESP32 package recently? If not, try to update it.

Of course, it can be something related to the the network equipment or settings as you mentioned.

This is the result I get (picture and text): 

https://imgur.com/a/pMxsx7M

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:808
load:0x40078000,len:6084
load:0x40080000,len:6696
entry 0x400802e4
ETH Started
ETH Connected
ETH MAC: 24:0A:C4:04:73:07, IPv4: 192.168.0.183, FULL_DUPLEX, 100Mbps

connecting to google.com
HTTP/1.0 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Tue, 02 Oct 2018 10:01:16 GMT
Expires: Thu, 01 Nov 2018 10:01:16 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
Age: 175
X-Cache: HIT from unconfigured
Via: 1.0 unconfigured:3128 (squid/2.6.STABLE1)
Proxy-Connection: close

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection
```

HarterHorst, 2018-10-02:
I can make it work and i can reproduce the error.

1. Upload script via USB (USB power Hub) - "Hard resetting via RTS pin..."
2. Open Serial Monitor. -> Not working - "ETH connected ... ETH disconnected".
3. Reset with button on PCB -> Not working - "ETH connected ... ETH disconnected".
4. unplug the usb cable. wait a few seconds and plug cable back in.
5. It Works!

The power off makes it work.

We can close this one here, because it's hardware related. But i will continue on olimex forum.

closed 2018-10-03 by HarterHorst

josepchetrit, 2021-08-02:
@HarterHorst 

Hi Harter,

I am trying your solution but I have a problem. When I unplug the USB cable and plug the cable back in, the serial on the monitor freezes and to see the monitor again I need to reload the script.

Thanks for your help