espressif/arduino-esp32 #1913: esp_ble_gap_set_prefer_conn_params not working as intended.

h2zero, 2018-10-01:
When using ESP32 as BLE client and calling esp_ble_gap_set_prefer_conn_params prior to connection it will connect with the specified parameters but immediately sends a esp_ble_gap_update_conn_params with  min_int=6, max_int=6, latency=0,timeout=600 right after connection,therefore changing the parameters sent with the call making it useless.

Arduino 1.8.5
arduino-esp32 core v1.0.0

This code will initially connect with a min and max interval of 50ms, 0 latency and 10 sec timeout.
Immediately after will be sent a params update with 7.5 min/max, 0 latency and 6 sec timeout, negating the intention.

```
bool connectToServer(BLEAddress pAddress) {
     BLEClient*  pClient  = BLEDevice::createClient();
     esp_bd_addr_t addr;
     memcpy(addr,pAddress.getNative(),6);
     esp_ble_gap_set_prefer_conn_params(addr, 0x0028,0x0028, 0, 0x03e8);
     pClient->connect(pAddress);
}
```

chegewara, 2018-10-01:
Did you try to change order and use it after pClient->connect()?

h2zero, 2018-10-02:
> Did you try to change order and use it after pClient->connect()?

Yes as a test I did try that, the result is the same. Intially connects with the requested parameters then immediately sends a parameter change.

The log on the peripheral looks like this:

```
Evt Connected 0 30:ae:a4:23:1a:5e
Params: 40 40 0 1000
enter sleep
exit sleep
Evt Conn Param Update 0x6 0x6 0x0 0x258
enter sleep
exit sleep
Evt Write, handle = 11
02 00 
enter sleep
exit sleep
Evt Conn Param Update 0x28 0x28 0x0 0x3E8
enter sleep
exit sleep
Evt Write, handle = 18
01 00 
enter sleep
exit sleep
Evt Write, handle = 15
01 00 
enter sleep

```

h2zero, 2018-10-02:
You are correct, it does eventually get there. But I believe this to be an unintended behavior that could be contributing to the occasional connection failures I see. 

No other client device I use does this so I consider it odd. I can't see how this would be intentional, but if it is I would be interested in knowing why.

h2zero, 2018-10-02:
I read your edit after I replied and considered the possibility that the peripheral wasn't agreeing with the client. So I set the parameters for both to the same defaults and the behavior is unchanged.  

I've also now tested this to the point that I'm certain it's causing a connection failure that seems to happen 1 out of 20 or so connections, as I have yet to see a failure without the call to esp_ble_gap_set_prefer_conn_params.

chegewara, 2018-10-02:
@h2zero i cant say 100%, but from my tests i can say that you are right. There is something wrong with it.
First of all when i call it before i connect to peer device then i cant establish connection:
```
esp_ble_gap_set_prefer_conn_params(addr, 0x0028,0x0028, 0, 0x03e8);
     pClient->connect(pAddress);
```
Im getting logs with mtu = 0 `BLEUtils: [status: ESP_GATT_ERROR, conn_id: 0, remote_bda: 70:85:4d:70:10:e1, mtu: 0]`.
When i change order and call it after i connect to peer device then there is nothing happen. I am trying to test it with esp-idf but im having some other issue. Im pretty sure its esp-idf issue or again i dont know how to use function :(

h2zero, 2018-10-02:
@chegewara thanks for trying it out, I was pretty certain there is a bug that needs addressed. I was just thinking about trying with the idf to see what happens, If you test it with the idf please post your results.

chegewara, 2018-10-02:
https://github.com/espressif/esp-idf/issues/2504

h2zero, 2018-10-02:
Actually, according to espressif's doc's its described as only used by the master, which is interesting because a master is described as the central device post start of connection and prior to connection completion. 

```
esp_err_t esp_ble_gap_set_prefer_conn_params(esp_bd_addr_t bd_addr, uint16_t min_conn_int, uint16_t max_conn_int, uint16_t slave_latency, uint16_t supervision_tout)

    This function is called to set the preferred connection parameters when default connection parameter is not desired before connecting. This API can only be used in the master role.
```  
Which is interesting, however I believe the intent is that the central/master device controls the connection parameters at all times and the peripheral just suggests its preference and the central/master makes the decision. 

In that scenario, should not the central first set the parameters to connect with and maintain those until a negotiated update of those parameters is desired or disconnection?

That's my understanding of the process and intent of the above function, please correct me if I am misunderstanding.

chegewara, 2018-10-02:
Yes, ive seen this description in documentation. It is possible that `esp_ble_gap_set_prefer_conn_params` and PPCP characteristic are two different things.
Maybe this can help you https://github.com/nkolban/esp32-snippets/issues/588

h2zero, 2018-10-05:
Thanks, I tried setting the parameters with the update param request function in the onConnect callback and the result is the same behavior. I believe something is happening at the lower levels in the IDF, I'll have to test it in the IDF and see what happens.

chegewara, 2018-10-15:
Hi @h2zero 
i dont know yet if it works in arduino-ide but in esp-idf master it seems to work. Here is my code:
```
       BLEAddress addr = *(BLEAddress*)data;
       BLEClient*  pClient  = BLEDevice::createClient();
       pClient->setClientCallbacks(new MyCallbacks());
       esp_ble_gap_set_prefer_conn_params(*addr.getNative(), 100,125, 2, 0x03e8);
       pClient->connect(addr);
```
and logs from nRF connect:
```
W	03:10:00.665	Connection parameters update failed with status 35 (interval: 156.25ms, latency: 2, timeout: 10000ms)
```

h2zero, 2018-10-17:
Thanks, I haven't had time to play with the idf yet but i'll give that a try when I can.

chegewara, 2018-10-18:
@h2zero This is just one commend call from idf you can use in arduino too. But there is something else, what could cause your issue. Even if you use this command properly, peer device can not accept parameters you are trying to use and will be used default 7.5ms. 
If you can just try different settings with nRF connect.

chegewara, 2018-11-29:
This function has been added in esp32-snippets now. It can be used in arduino.

h2zero, 2019-04-01:
Totally forgot about this, works correctly in latest release.

closed 2019-04-01 by h2zero