espressif/arduino-esp32 #1914: Problem with LEDControl.h

ShoootLight, 2018-10-01:
### Hardware:
Board:							NodeMCU ESP-32S
Core Installation/update date:			28/sep/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                        no
Upload Speed:					         921600
Computer OS:                                            Windows 10

### Description:
For a school project I need to control 8 segments of the MAX7219 using the [LEDControl ](https://playground.arduino.cc/Main/LedControl)library. A half year ago when I paused the project everything worked but two days ago when I started again, it did nothing. After uploading the sketch the matrix lights completely up (every pixel) but when I replug it into my computer nothing happens. But the same works perfectly when I use an Arduino Uno. The debug message is the same as when I upload other sketches. I tried reinstalling the Arduino IDE and both ways ([Board Manager](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md) and [git GUI](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md)) of installing the Arduino Core, also the [avr/pgmspace.h fix](https://github.com/espressif/arduino-esp32/issues/508). I tried different Pin configurations (GPIO Pin 4, 2, 0 (used back in the days) and 13, 10, 9). Everything with no success.

I say thank you in advance for your time. 
Yours sincerely
Stephan P.

### Sketch:  (It's unnecessary, it should just show that controlling isn't possible, thanks to [Brainy Bits ](https://www.brainy-bits.com/) for their [sketch](https://www.brainy-bits.com/how-to-control-max7219-led-matrix/))
```cpp

//This sketch belongs to the guys from Brainy Bits (https://www.brainy-bits.com/how-to-control-max7219-led-matrix/)
#include "LedControl.h"

LedControl lc=LedControl(12,10,11,2);  // Pins: DIN,CLK,CS, # of Display connected

unsigned long delayTime=200;  // Delay between Frames

// Put values in arrays
byte invader1a[] =
{
   B00011000,  // First frame of invader #1
   B00111100,
   B01111110,
   B11011011,
   B11111111,
   B00100100,
   B01011010,
   B10100101
};

byte invader1b[] =
{
  B00011000, // Second frame of invader #1
  B00111100,
  B01111110,
  B11011011,
  B11111111,
  B00100100,
  B01011010,
  B01000010
};

byte invader2a[] =
{
  B00100100, // First frame of invader #2
  B00100100,
  B01111110,
  B11011011,
  B11111111,
  B11111111,
  B10100101,
  B00100100
};

byte invader2b[] =
{
  B00100100, // Second frame of invader #2
  B10100101,
  B11111111,
  B11011011,
  B11111111,
  B01111110,
  B00100100,
  B01000010
};

void setup()
{
  lc.shutdown(0,false);  // Wake up displays
  lc.shutdown(1,false);
  lc.setIntensity(0,5);  // Set intensity levels
  lc.setIntensity(1,5);
  lc.clearDisplay(0);  // Clear Displays
  lc.clearDisplay(1);
}

//  Take values in Arrays and Display them
void sinvader1a()
{
  for (int i = 0; i < 8; i++)  
  {
    lc.setRow(0,i,invader1a[i]);
  }
}

void sinvader1b()
{
  for (int i = 0; i < 8; i++)
  {
    lc.setRow(0,i,invader1b[i]);
  }
}

void sinvader2a()
{
  for (int i = 0; i < 8; i++)
  {
    lc.setRow(1,i,invader2a[i]);
  }
}

void sinvader2b()
{
  for (int i = 0; i < 8; i++)
  {
    lc.setRow(1,i,invader2b[i]);
  }
}

void loop()
{
// Put #1 frame on both Display
    sinvader1a();
    delay(delayTime);
    sinvader2a();
    delay(delayTime);

// Put #2 frame on both Display
    sinvader1b();
    delay(delayTime);
    sinvader2b();
    delay(delayTime);

}
```

stickbreaker, 2018-10-01:
The pins you assigned as controlling the led cannot be used.  Specifically:
```c++
LedControl lc=LedControl(12,10,11,2);  // Pins: DIN,CLK,CS, # of Display connected
```

Chuck.

MarkyAD, 2018-10-01:
And pins 0, 2, 12 are usable only if certain conditions are met. pins 2 and 12 can't be pulled high at boot and pin 0 can't be pulled low at boot, otherwise they'll interfere with booting and/or flashing.

beegee-tokyo, 2018-10-01:
[Reserved GPIO’S](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

ShoootLight, 2018-10-01:
First of all thank you all for your quick reply. I tested it with other GPIO pins with the result that the matrices are now flickering but I think I messed up my board (?) with pin 2, 12 and 0

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57
`
(repeatedly shown in serial monitor, instead of things it should show)

so I'm going to flash the firmware and retry it. I write the results and possibly close the thread.

ShoootLight, 2018-10-07:
I now used P32, 33 and 25. And it works :D! Thank you all for your support, especially @chegewara !

ShoootLight, 2018-10-07:
I close  #1914

closed 2018-10-07 by ShoootLight