espressif/arduino-esp32 #1915: Use PSRAM into WiFi, BLE, HTTP, MBEDTLS(SSL)

ArTyGill, 2018-10-01:
### Description:

How to use psRam for wifi (ssl) and BLE in Arduino ide?

Just turning on the psram does not do anything because you need to use ps_malloc () to use external memory, how can I fix this?

beegee-tokyo, 2018-10-01:
Not sure about ArduinoIDE, but in PlatformIO I use ESP32-Wrover board as target and BT uses PSRam automatically. Was able to run BTserial, Wifi in AP mode, Wifi async webserver in parallel without problems.

ArTyGill, 2018-10-01:
So you tried to run wifi http, bt and sd card, but did it work?

beegee-tokyo, 2018-10-01:
I use SPIFFS for the async webserver files, but yes, it works. BT used around 40k less RAM after I chose Wrover board. That was good enough to get the rest to work. WiFi doesnt show any improvement in heap usage.

beegee-tokyo, 2018-10-01:
Heap usage (in my specific project) with PSRAM:
```
[HEA] ##################################
[HEA] on start: 
[HEA] Internal Total heap 230952, internal Free Heap 201412
[HEA] SPIRam Total heap 4194252, SPIRam Free Heap 4194252
[HEA] ChipRevision 1, Cpu Freq 240, SDK Version v3.2-dev-1055-g3276a1316
[HEA] Flash Size 4194304, Flash Speed 40000000
[HEA] ##################################
[HEA] after readPreferences: 201196
[HEA] after initBTSerial: 150492
[HEA] after startDisplay: 147484
[HEA] after initOTA: 108824
[HEA] after WiFi.begin(): 107792
[HEA] after SPIFFS begin: 103456
[HEA] after startWebServices: 93596
[HEA] after initWiFi: 88912
```
and without PSRAM (**_webserver is not started here because there is not enough free heap. System would just crash_**):
```
[HEA] ##################################
[HEA] on start: 
[HEA] Internal Total heap 230984, internal Free Heap 201560
[HEA] SPIRam Total heap 0, SPIRam Free Heap 0
[HEA] ChipRevision 0, Cpu Freq 240, SDK Version v3.2-dev-1055-g3276a1316
[HEA] Flash Size 4194304, Flash Speed 40000000
[HEA] ##################################
[HEA] after readPreferences: 201344
[HEA] after initBTSerial: 102612
[HEA] after startDisplay: 103400
[HEA] after initOTA: 59356
[HEA] after WiFi.begin(): 60476
[HEA] after initWiFi: 56088

```

ArTyGill, 2018-10-01:
Really on platforms Io, it is spent less memory ?? On Arduino it just disappears when adding modules!

beegee-tokyo, 2018-10-01:
The above logs are taken with the same app running on an ESP32 Wrover and an ESP32 Wroom, both  programmed from PlatformIO with the following platformio.ini:
```
[env:remote-smart]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp-wrover-kit

framework = arduino

; upload_port = 192.168.0.111
upload_port = COM15
upload_speed = 921600

board_build.partitions = B:\Projects\ScentControl\Remote_Smart_II\custompart.csv

build_flags =
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-Os
	-fmerge-all-constants
	-ffast-math
	-fira-loop-pressure
	-fdata-sections
	-fexceptions
	-DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG
	-DSW_VERSION=0
	-DSW_SUBVERSION=0
	-DSW_MINVERSION=1
	-DFS_VERSION=0
	-DFS_SUBVERSION=0
	-DFS_MINVERSION=1

[platformio]
data_dir = ./src/data
```
I have no reason to serve false information here.
Be aware that I use the dev version of Arduiino-ESP32, not the official release. There might be differences in the use of PSRam between them.

lbernstone, 2018-10-01:
Most of these depend on the underlying esp-idf components being able to use PSRAM.  That is happening- bluedroid and lwip made it into the current build of arduino-esp32; mbedtls has been updated in the last week.  It would be good to identify which arduino-esp32 components are using large memory allocations and make those psram aware.

ArTyGill, 2018-10-02:
when trying to start client http, an error occurs: rsa key bignum memory allocation error

ArTyGill, 2018-10-02:
same error as in this issue

https://github.com/espressif/arduino-esp32/issues/1420

ArTyGill, 2018-10-03:
Free rtos used a lot of RAM, after disabling some tasks. I was able to run bluetooth and wifi at the same time.

ArTyGill, 2018-10-04:
It would be great if the stack of a free real-time operating system was loaded into psram via iram, and did not take the place of iram.

closed 2018-10-04 by ArTyGill