espressif/arduino-esp32 #1918: DevKitc-V4 with ESP32-SOLO-1 not supported

beegee-tokyo, 2018-10-02:
How can I get the ESP32-SOLO-1 on an original Espressif DevKitc-V4 working with this framework?

Is it required to create a now board description for it?

If yes, is there any guidance how to do it?

[ESP32-SOLO-1](https://docs.espressif.com/projects/esp-idf/en/latest/hw-reference/modules-and-boards.html#esp32-solo-1) is a single core 160MHz variant of the ESP32.

beegee-tokyo, 2018-10-03:
@QuadCorei8085 
Talked with Me-No-Dev on Gitter.
Support for the SOLO-1 is on the todo list, but no date for availability. So for now we can program the SOLO-1 only with ESP-IDF.

QuadCorei8085, 2018-10-22:
Is there maybe a workaround we could apply locally?

I tried to modify the SDK config by adding the define "CONFIG_FREERTOS_UNICORE" but compiled app did not work with ESP32 SOLO.

beegee-tokyo, 2018-10-22:
@QuadCorei8085 You can try to use Arduino as a component in ESP-IDF. Its on my todo-list, but didn't find time yet.

lbernstone, 2019-04-13:
https://github.com/lbernstone/arduino-esp32-solo

closed 2019-08-15 by stale[bot]