espressif/arduino-esp32 #1920: BLE SERVER- CAN NOTIFY TO ONLY ONE CLIENT?

sizen801, 2018-10-03:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do.
Describe your system( Hardware, computer, O/S, core version, environment)
Describe what is failing
Show the shortest possible code that will duplicate the error
Show the EXACT error message(it doesn't work is not enough)
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                          ?no?
Upload Speed:						?115200?
Computer OS:                                            ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
@

chegewara, 2018-10-03:
Original ble library for arduino is not yet ready to work with multiple connection, server to multiple clients or client to multiple servers.

beegee-tokyo, 2018-10-10:
@chegewara 
I have the opposite problem, One client connecting to two servers. Registering on both servers for notifications, but only last registered gets notified. First registration is lost.
If you are working on something to enable multiple notifications and have some code to test, I would like to test it.

chegewara, 2018-10-10:
I am currently working on few upgrades/enhancements in library. This is on my todo list. I already have working code one esp32 server to many clients, just need some more tests.

beegee-tokyo, 2018-10-10:
I need the other way, one client to multiple servers :-(
Will try to patch BLERemoteCharacteristic.cpp by myself. and share with you if I get it to work. Are you available on Gitter for discussion?

chegewara, 2018-10-10:
I am now.

chegewara, 2018-11-29:
It should work now.

closed 2019-08-15 by stale[bot]