espressif/arduino-esp32 #1922: HTTPClient: getStream() doesn't work if _secure=true

ranr, 2018-10-03:
Hi,

getStream() when using SSL returns an error (by checking !_secure flag):

```cpp
WiFiClient& HTTPClient::getStream(void)
{
    if (connected() && !_secure) {
        return *_tcp;
    }

    log_w("getStream: not connected");
    static WiFiClient empty;
    return empty;
}
```
Surprisingly the getStreamPtr function doesn't have that restriction. 
Looking at the annotation I traced the commit where that condition was added. This is commit cfbb7300b768a1ecd7f1300ecac044268153e78b

Commit note says: 
Ensure errors are generated if the wrong method is called to retrieve the SSL data stream. I couldn't find why there would be a difference between these two functions. Also when I remove the restriction and use getStream it works. Googling it didn't give any result either.

@lonerzzz and @me-no-dev - could you please help shed some light on this? (as it's your commit)

lonerzzz, 2019-08-01:
This issue should not be marked stale as the OP's question has not been answered.

atanisoft, 2019-08-01:
@lonerzzz this was fixed in https://github.com/espressif/arduino-esp32/commit/01d22c8807cf9186e242c7c64198d67e88939db9 ... this issue should get closed...

@me-no-dev FYI...

ChrisBeeson, 2019-10-08:
When I use BearSSL .setInsecure() and getStream() the bytes downloaded are not as expected.

```
  HTTPClient https;
  BearSSL::WiFiClientSecure secureClient;
  secureClient.setInsecure();
  https.begin(secureClient, STORAGE_BASE_URL, 443, url, true);
  WiFiClient stream = https.getStream();
```
The first 32 bytes downloaded from a firebase bucket are:

`0000 17 03 03 03 e7 89 f2 6a cc dd 73 1c 86 46 6b 4c`

But if I download the file in my browser the bytes are:
`0000 e9 01 02 40 9c f2 10 40 00 f0 10 40 68 05 00 00`

Any ideas?

me-no-dev, 2019-10-08:
@ChrisBeeson seems you are talking about esp8266? what BearSSL?

closed 2019-10-08 by me-no-dev

Hassaan18, 2021-10-01:
@ChrisBeeson How'd you be able to display these bytes?