espressif/arduino-esp32 #1924: WiFi.softAP("apname", "") not working

jeroenst, 2018-10-03:
WiFi.softAP only works when passphrase is 0, in esp8266 it also works when passphrase is "".

In WiFiAP.cpp on line 104 should change from:

    if(passphrase && (strlen(passphrase) > 63 || strlen(passphrase) < 8)) {

to:

    if(passphrase && (strlen(passphrase) > 63 || strlen(passphrase) < 8 || strlen(passphrase) == 0)) {

MarkyAD, 2018-10-04:
>WiFi.softAP only works when passphrase is 0

What you probably mean is "WiFi.softAP() only works when the length of passphrase is >= 8 characters, which is true. The whole routine fails if too long, too short, or no passphrase is given. It also fails if ssid is too long.

> if(passphrase && (strlen(passphrase) > 63 || strlen(passphrase) < 8 || strlen(passphrase) == 0)) {

This doesn't help much because its already testing for strlen(passphrase) < 8.

What would make this routine more robust would be:
- If passphrase or ssid is too long, truncate them to an appropriate length.
- If passphrase is too short or empty, fall back to using WIFI_AUTH_OPEN like it does about 12 lines down where it sets conf.ap.authmode.
- fail only if ssid is not given or AP not enabled.

MarkyAD, 2018-10-04:
@jeroenst you are correct that 0 length passphrases were failing when they should have just set an open AP. I put up a pull request that should fix it. In the meantime you can try replacing the one line with the following:
```c++
if(passphrase && strlen(passphrase) > 0 && (strlen(passphrase) > 63 || strlen(passphrase) < 8)) {

jeroenst, 2018-10-04:
You're right, my proposed fix was wrong. Thanx for the solution!

jeroenst, 2018-10-04:
Tested your fix, and it accepts passphrase "" now.

MarkyAD, 2018-10-04:
Yes it should have been checking strlen(passphrase) > 0 first. Please close if this solves the issue.

jeroenst, 2018-10-04:
Shouldn't I wait with closing until this issue has been solved in the master branch?

closed 2018-11-19 by me-no-dev

solroshan, 2020-12-21:
I got same error on my serial monitor. Compiler builds with any error but serial monitor prints;
**[E][WiFiSTA.cpp:219] begin(): connect failed! [E][WiFiAP.cpp:109] softAP(): passphrase too short!**
 How can I rearrange passphrase as ">8" ?

PR #1925 (2018-11-19): make WiFi.softAP() more robust
Make WiFi.softAP() fail less easily, attempt to fix #1924