espressif/arduino-esp32 #1926: mDNS causes assertion when switching for STA to AP mode

jeroenst, 2018-10-04:
When switching from station mode to access point mode the esp32 crashes with the following error:

'assertion "igmp_lookup_group: first group must be allsystems" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/igmp.c", line 269, function: igmp_lookup_group
abort() was called at PC 0x400dbd03 on core 1

Backtrace: 0x4008f49c:0x3ffb4ee0 0x4008f671:0x3ffb4f00 0x400dbd03:0x3ffb4f20 0x400fe0b2:0x3ffb4f50 0x400fe296:0x3ffb4f70 0x400da3be:0x3ffb4f90 0x400da52c:0x3ffb4fd0 0x400da58d:0x3ffb4ff0 0x400f3828:0x3ffb5010 0x4008b525:0x3ffb5040

Rebooting...'

When enabling verbose logging, the error doesn't occur:

'rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1496
load:0x40078000,len:8596
load:0x40080400,len:6980
entry 0x400806f4
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 14 - AP_STOP
[E][WiFiSTA.cpp:211] begin(): connect failed!
[D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
'

Code I used to reproduce this bug:

`#include <WiFi.h>
#include <WiFiUdp.h>
#include <ESPmDNS.h>

void setup() 
{  
  MDNS.begin("esp32");
  WiFi.mode(WIFI_STA);
  WiFi.begin();
  WiFi.softAP("HELLO_WORLD","",6,0);
}

void loop() {
  // put your main code here, to run repeatedly:

}
`

MarkyAD, 2018-10-04:
Hello @jeroenst see #1885 for possible workaround.

MarkyAD, 2018-10-04:
Issue was for mDNS + OTA but sounds related.

MarkyAD, 2018-10-04:
@jeroenst maybe try not using the WiFi.mode() and WiFi.begin(); lines, also try configuring AP before calling WiFi.softAP() like...
```c++
const IPAddress ap_IP = IPAddress(192, 168, 4, 1);   // ip address
const IPAddress ap_NM = IPAddress(255, 255, 255, 0); // netmask

WiFi.softAPConfig(ap_IP, ap_IP, ap_NM);
WiFi.softAP("HELLO_WORLD", "",6,0);

MarkyAD, 2018-10-04:
@jeroenst I tried your exact code with several logging levels and couldn't reproduce it. @beegee-tokyo was saying his changed depending on the ordering of various function calls. If youre using the git version of the core it may be worth it to checkout an older git version (say before the last idf update commit #1878) or to the release version and see if it still does it.

jeroenst, 2018-10-05:
I think this is allready fixed in the master branch and I was using the esp32 downloaded by the boards manager of arduino.

I removed the esp32 from the boards manager and cloned the master branch into Arduino/hardware, now I don't get the error anymore.

Sorry for misusing your time.

closed 2018-10-05 by jeroenst

qt1, 2018-10-10:
Same here. happens in MDNS.begin ..

Aietes, 2018-10-20:
I don't think this is fixed, getting the same error with current branch.

fabiuz7, 2018-11-22:
I can confirm: I've get the same message error when mDNS is initialized. This error seems to be sporadic, but usual happens after button reset. My code is still messy, I going to investigate to provide more info...

luc-github, 2018-11-22:
I have also randomly the assertion issue - I was hoping this commit will fix it 
https://github.com/espressif/esp-idf/commit/43936ab48acb0df5eadb309d66b7357775524338
could be ?