espressif/arduino-esp32 #1931: SimpleWifiServer only one client at the same time?

provotector, 2018-10-05:
### Hardware:
Board:							ESP32 Dev Module. Also tested in Wemos LOLIN 32
Core Installation/update date:	       12/aug/2018
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      NO
Upload Speed:				        115200
Computer OS:                                          Windows 10

### Sketch: SimpleWifiServer Example from Arduino IDE

### Description:

Sometimes the client is not "flushed" and the webserver can not serve pages. 

The normal behavior at serial monitor is:

```
New Client.
GET /favicon.ico HTTP/1.1
Host: 192.168.0.164
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36
Accept: image/webp,image/apng,image/*,*/*;q=0.8
Referer: http://192.168.0.164/
Accept-Encoding: gzip, deflate
Accept-Language: es-ES,es;q=0.9,en-US;q=0.8,en;q=0.7

Client Disconnected.
```

But sometimes, when a client refresh or interacts with the webpage the program flow stops at "New client". The webpage is not loading and the client is not flushed. At this point the webpage can not be reached from another client isntead the client caused the "hang" refreshing the page.

Serial monitor output. Please pay attention at the last line. A new client is detected but the program flow stops.

```
New Client.
GET /favicon.ico HTTP/1.1
Host: 192.168.0.164
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/69.0.3497.100 Safari/537.36
Accept: image/webp,image/apng,image/*,*/*;q=0.8
Referer: http://192.168.0.164/
Accept-Encoding: gzip, deflate
Accept-Language: es-ES,es;q=0.9,en-US;q=0.8,en;q=0.7

Client Disconnected.
New Client.      <------- The sketch "hangs" at this point
```
After waiting a few minutes, the client is disconnectec and I can see this at serial monitor. I was configured the debug level to "verbose".

```
[D][WiFiClient.cpp:452] connected(): Disconnected: RES: 0, ERR: 128
Client Disconnected.
```

provotector, 2018-10-16:
No updates? No answers? The "SimpleWifiServer.ino" is a bugged example? Some help please. Thanks nae9 for sharing your info.

poengAlex, 2018-10-25:
Same problem here. I have the ESPAsyncWebServer library installed. Do they interferer?

nae9, 2018-10-25:
May be problem in wificlient.cpp in client.stop() function - they not work in such manner as early version.

poengAlex, 2018-10-25:
I rerolled my code to see the difference. One change in a macro made the error disappear.
From:
#define TIMEMS micros()
TO:
#define TIMEMS system_get_time() / 1000 

I use TIMEMS a lot of places so I guess that could be anything else. Or it might be that the Arduino micros() function mess things up. I will try to find time to some more debugging.

lbernstone, 2018-10-25:
Does micros() consistently give the same result as system_get_time()?  system_get_time is marked as deprecated.

closed 2019-08-15 by stale[bot]