espressif/arduino-esp32 #1933: This has Windows trojans.

muchginger, 2018-10-06:
---ClamScan report---
/home/<USER>/arduino-1.8.5/hardware/espressif/arduino-esp32/tools/espota.exe: Win.Dropper.Occamy-6651542-0 FOUND
/home/<USER>/arduino-1.8.5/hardware/espressif/arduino-esp32/tools/get.exe: Win.Dropper.Occamy-6651542-0 FOUND
/home/<USER>/arduino-1.8.5/hardware/espressif/arduino-esp32/tools/gen_esp32part.exe: Win.Dropper.Occamy-6651542-0 FOUND

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			12/Dec/2017
IDE name:						Arduino IDE
Computer OS:                                            Ubuntu

m-menon, 2018-10-08:
The same "Win.Dropper.Occamy-6651542-0" detected in the get.exe that I downloaded. Anyone care to explain this?

koskee, 2018-10-09:
Did you download it from this repository or elsewhere? Beware other sites that may claim to offer downloads of free software. Best to download from the official site whenever possible (like this page).

That being said, no antivirus software is accurate 100% of the time, as they often employ predictive algorithms which is not dissimilar to behavior of AI engines/algorithms. Thus it would be nearly impossible to always be correct, as learning is a major part of how it makes decisions.

Id call it a false positive, personally.

m-menon, 2018-10-09:
@koskee thanks for clarifying. I downloaded from this repo - absolutely! :)

As you mentioned I am starting to think that the AV got triggered on false alarm ( cos "get.exe" when executed downloads files from the web - hence looked like some dropper? ). 

Bitdefender didn't sound an alarm whereas ClamAV did. But I think that's just a false alarm - and from what I read earlier, get.exe is nothing but a windows executable version of get.py. Those paranoid ( occasionally me too :) ) can perhaps run python get.py.

koskee, 2018-10-09:
Yeah, since Python isn't compiled, you'd be able to directly read the code if you were suspicious about what the script was doing. 

But since these tools have been used by a ton of people without issue (I assume) it should be fine.

m-menon, 2018-10-09:
thanks @koskee

closed 2019-08-15 by stale[bot]