espressif/arduino-esp32 #1937: How to find IP address of ESP32?

nomanshahsyed, 2018-10-08:
I need to know the IP address of ESP32 on the local network. The idea is to do mDNS or UDP broadcast to send the IP to the android application. The app will then use that IP to do the communication. Is there someone who has already done it? Thanks.

martinius96, 2018-10-08:
´WiFi.localIP();´
use it with serial print for instance:
´Serial.println(WiFi.localIP());´

nomanshahsyed, 2018-10-08:
Hi martin thanks for reply. You didn't get my point. I need to know the IP address without serial monitor. I have tried mDNS which works fine but not on Android. I am now trying UDP with which I have some success.

uzi18, 2018-10-08:
WiFi.localIP(); returns IPAddress, You can print it or use it somewhere.

martinius96, 2018-10-08:
@nomanshahsyed set it static then :)
https://github.com/espressif/arduino-esp32/issues/1262#issuecomment-386448251

nomanshahsyed, 2018-10-09:
Guys thanks for your answers. But please read the question before answering. Anyhow the answer to my question is to use UDP broadcast. mDNS is also working but not on android. UDP broadcast is working fine.

closed 2018-10-09 by nomanshahsyed

chuledeco, 2019-12-26:
> Guys thanks for your answers. But please read the question before answering. Anyhow the answer to my question is to use UDP broadcast. mDNS is also working but not on android. UDP broadcast is working fine.

Can you please put here how did you solve this problem? Thanks

uzairali001, 2020-05-08:
I also want to achieve this, can you please share the details how you managed to do it?

martinius96, 2020-05-08:
mDNS must be enabled on devices and also on network... 
It is not that simple like.. I will start it on ESP and it will work on each device. You need to ALLOW Multicast DNS in your network and on your devices aswell..