espressif/arduino-esp32 #1939: MLX90614 some time not giving right values on i2c

jjassar, 2018-10-08:
Hello all,
I Need help regarding MLX90614 i2c communication.

I am using a ESP32 board which is communicating with MLX90614 i2c temperature sensor.
Sensor is directly soldered on the ESP32 board, with 4.7k pullups.
I am using ESP32 release 1.0.0, Library used to communicate with MLX90614 is 
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library

Default Release 1.0.0 gives occasionally error in reading values 
Then i replaced 4 files from latest GitHub Repo, as guided @stickbreaker, after which reading from sensor become more stable.

But once it start giving wrong temperature values then wrong values comes continuously.
Problem is solved most of time by calling ESP.restart()
But sometimes only hardware restart solves the problem.

Also to note that when i add transact() methods to wire library created by @stickbreaker and calls transact() in Sparkfun Library then the error rate reduces.

Please help how can i make reading more stable.

stickbreaker, 2018-10-08:
@jjassar more info is needed.
* smallest code that will generate errors
* hardware description(schematic)
* exact error output
* set VERBOSE error reporting
* enable i2c advanced debug by uncommenting
 #define  ENABLE_I2C_DEBUG_BUFFER
In esp-hal-i2c.c

Chuck.

jjassar, 2018-10-11:
Hello @stickbreaker, sorry for delay. I am on leave due to family emergency. Will update soon as I get back to work. 
Thanks a lot