espressif/arduino-esp32 #1941: SSID is hide after set esp_wifi_set_protocol( WIFI_IF_AP, WIFI_PROTOCOL_LR )

NickChungVietNam, 2018-10-09:
**Hardware : 
ESP32-WROOM.
Ide :1.8.2.**

Hi Sir.
After try this code, my ESP32 hided SSID of AP forever. I try again with my second esp32 , the result is both 2 esp32 is hide SSID forever, how can i fix to show SSID again?
This is the code 'killed' my esp32 : (dont try to fash your ESP this code) 

```ruby
#include <Arduino.h>
#include <WiFi.h>
#include <WiFiUdp.h>

#include <esp_wifi.h>

const char *ssid ="ssid";
char *pass="123456789";
WiFiUDP udp;

const char *toStr( wl_status_t status ) {
switch( status ) {
case WL_NO_SHIELD: return "No shield";
case WL_IDLE_STATUS: return "Idle status";
case WL_NO_SSID_AVAIL: return "No SSID avail";
case WL_SCAN_COMPLETED: return "Scan compleded";
case WL_CONNECTED: return "Connected";
case WL_CONNECT_FAILED: return "Failed";
case WL_CONNECTION_LOST: return "Connection lost";
case WL_DISCONNECTED: return "Disconnected";
}
return "Unknown";
}

void setupAp() {
WiFi.begin();
delay( 500 ); // If not used, somethimes following command fails
WiFi.mode( WIFI_AP );
ESP_ERROR_CHECK( esp_wifi_set_protocol( WIFI_IF_AP, WIFI_PROTOCOL_LR ) );

Serial.println( WiFi.softAP( ssid, pass ) );
Serial.println( WiFi.softAPIP() );
delay( 1000 );
}

void setup() {
Serial.begin( 115200 );
Serial.println( "Master" );
setupAp();
udp.begin( 8888 );
}

void loop() {
udp.beginPacket( { 192, 168, 4, 255 }, 8888 );
udp.write( 'b' );

if ( !udp.endPacket() )
    ESP.restart(); // When the connection is bad, the TCP stack refuses to work
delay( 1000 );
}
```

beegee-tokyo, 2018-10-09:
I think your problem is that you set your WiFi to **`WIFI_PROTOCOL_LR`**.
While in Long Range mode, the WiFi cannot be seen by "normal" WiFi devices. Try to erase NVS partition of your ESP32 and / or flash a firmware that does NOT set the WiFi into Long Range mode. I was playing around with Long Range mode before and could get back to normal WiFi mode.
For erasing NVS partition check out [WiFi connection problem - NVS corrupted](https://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/) for a code snippet how to erase the NVS partition.

NickChungVietNam, 2018-10-09:
> Tôi nghĩ rằng vấn đề của bạn là bạn thiết lập WiFi của bạn **`WIFI_PROTOCOL_LR`**. 
> Trong khi ở chế độ Long Range, không thể nhìn thấy WiFi bằng thiết bị WiFi "bình thường". Cố gắng xóa phân vùng NVS của ESP32 và / hoặc flash firmware mà KHÔNG đặt WiFi thành chế độ Long Range. Tôi đã chơi xung quanh với chế độ Long Range trước đây và có thể quay lại chế độ WiFi bình thường. 
> Để xóa phân vùng NVS, hãy kiểm tra sự [cố kết nối WiFi - NVS bị hỏng](https://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/) cho một đoạn mã làm thế nào để xóa phân vùng NVS.

closed 2018-10-09 by NickChungVietNam

NickChungVietNam, 2018-10-09:
**beegee-tokyo ** You are my god.  Thank you so much.  
I copy the code fix on this page : [https://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/](WiFi connection problem - NVS corrupted ) 
WiFi connection problem – NVS corrupted
If you have WiFi connection problems where WiFi.status() always returns WL_NO_SSID_AVAIL even you are sure that the AP is available and working, it might be that the NVS storage has been corrupted.

The NVS storage is the place where esp-idf stores the WiFi credentials. See issue wifi connection problem on Arduino-esp32. To check if your NVS storage is corrupted you can use ESP32-Show_nvs_keys provided by stickbreaker.

If your NVS storage is corrupted you can use the following code snippet to wipe all stored NVS keys. Attention, you will loose your previously stored WiFi credentials when using this.
This the code i used.
```` ruby
#include <nvs.h>
#include <nvs_flash.h>
void clearNVS() {
    int err;
    err=nvs_flash_init();
    Serial.println("nvs_flash_init: " + err);
    err=nvs_flash_erase();
    Serial.println("nvs_flash_erase: " + err);
 }
**beegee-tokyo ** You are my god. 
void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
clearNVS();
}

void loop() {
  // put your main code here, to run repeatedly:

}
````

beegee-tokyo, 2018-10-09:
Happy I could help.