espressif/arduino-esp32 #1942: MCP23S17 for ESP32

kennythum, 2018-10-09:
Hello all! I'd been searching the whole internet for several days, looking for a working library to work with MCP23S17 SPI I/O Extender. Does anyone have any experience on using this chip?? I tried with some existing library which uses for arduino, but it seems like does not working. Please help!

Guillaume57, 2019-04-18:
Hello, I just wanted to say that I got it working by doing this:
```#include <SPI.h>
#include <MCP23S17.h> // https://github.com/MajenkoLibraries/MCP23S17

SPIClass hspi( HSPI );
MCP23S17 mcp23s17( &hspi, 15, 0 );```

Connect GPIO12 to SO, GPIO13 to SI, GPIO14 to SCK, GPIO15 to CS

Mesihas, 2019-06-22:
Hello Guillaume57, I am trying to have working a tft/touch/sd card which uses SPI library and by default uses VSPI bus. According to Expressif you can connect up to 3 devices per bus in ESP32 so it means I that if I want to connect a MCP23S17..I have to use HSPI as I reached the maximum number of devices for VSPI. Any idea about my scenario?
Cheers

closed 2019-09-04 by stale[bot]