espressif/arduino-esp32 #1944: send multiple variables using one http.POST request ESP32

anandpanchal, 2018-10-09:
### Hardware:
Board:							        ESP32 Dev Module
Core Installation/update date:			2018
IDE name:							Arduino IDE
Flash Frequency:					        80Mhz
PSRAM enabled:                                             no
Upload Speed:					        115200
Computer OS:                                                 Windows 7

### Description:
Hello Everyone,

i'm trying to send(POST) multiple(2or3) variables using http.POST , but unable to do so. can anyone please help me in doing this ?

Thanks
Anand

[image]

chegewara, 2018-10-09:
You can pass one long string which is in form of:
```
grant_type=refresh_token&client_id=amzn1.application-oa2-client&refresh_token=Atzr%7CIwEBID5spY2iwIvtUw5sBKwzHSML
```
But you have to set header:
`Content-Type: application/x-www-form-urlencoded`

anandpanchal, 2018-10-09:
thanks for the replay.
i do not want to send the url. i need to send date and time. e.g http.post(date, time) something like this

anandpanchal, 2018-10-09:
http.addHeader("Content-Type", "application/x-www-form-urlencoded");
httpResponseCode = http.POST(date=dayStamp&timeis=timeStamp);
format of dayStamp = 08:09:2018
timeStamp = 12:14:34

error message : no match for 'operator&' (operand types are 'String' and 'String')

chegewara, 2018-10-09:
`httpResponseCode = http.POST("date=dayStamp&timeis=timeStamp");`

anandpanchal, 2018-10-09:
[image]
i am getting this on server side. but i want something like 08:11:2018 12:23:45

chegewara, 2018-10-09:
I understand, i just gave you example how to fix that error. Now you have to build string before you pass it as argument. Something like that:
```
char _post[40];
sprintf(_post, "date=%s&timeis=%s", dayStamp, timeStamp);
httpResponseCode = http.POST(_post);
```
Just check how to convert dayStamp and timeStamp to string (google it).

closed 2019-08-15 by stale[bot]

mehrdad987, 2019-08-16:
hey
Is there any restriction on sending data?
I mean I am sending a file content as one string and after 210 lines it couldn't send
whats a problem?