espressif/arduino-esp32 #1946: Guys can we have both AP and StA mode with Wifiserver bound to only AP ip address?

danbicks, 2018-10-09:
Currently I am tinkering with ESP32 boards and want to get up and running with what should be a simple implementation, but it seems that I am missing some key parameters.

The overview:

Various clients must connect to a core module using UDP or TCP but must connect and communicate only with the core device through AP connection 192.168.4.1 etc.

The core module will receive information from the clients and push this data to node red in the cloud through standard internet wifi connection "STA" mode.

The problem that I find is this.

If wifi mode is set to APSTA the TCP/UDP server instance seem to bind to STA rather than AP so communication from the clients is not received but only received through the STA IP and not when connected to the AP.

If I configure the mode for AP only, clients connect and send data without any problems.

Is there a way to bind a TCP / UDP server instance to only the AP IP address and allow STA to still communicate normally to the cloud?

Any input on this would be awesome.

Big thanks

Dans

danbicks, 2018-10-09:
`#include <WiFi.h>
#include <WiFiUdp.h>
#include <WiFiAP.h>
#include "OLED.h"

    
// WiFi network name and password:
const char * networkName = "mysidname";
const char * networkPswd = "routerpassword";
#define AP_SSID  "ESP32_SERVER"
const char* asecret = "12345678";
// Access Point IP Bindings
IPAddress ip( 192, 168, 4, 1 );   
IPAddress gateway( 192, 168, 4, 1 );  
IPAddress subnet( 255, 255, 255, 0 );
            
const int TcpServerPort = 1234;
const int TcpClientPort = 5234;
String TCPInput = ""; // string for input

 // Use WiFiClient class to create TCP connections
WiFiServer TCPServer(TcpServerPort,ip); // Does not bind to AP IP?
WiFiClient TCPClient;
#include "TCPSERVER.h"

//Are we currently connected?
boolean connected = false;

void setup(){
  // Initilize hardware serial:
  Serial.begin(115200);
  Serial.println("TCP Server Application Started..");
  Init_Display();
  //Connect to the WiFi network
  connectToWiFi(networkName, networkPswd, true); // connect to wifi and enable AP
  //connectToWiFi(networkName, networkPswd, false); // Only access point
}

void loop(){

  // Handle TCP here
   Handle_TCP();// Added TCP listener 
 
  
  delay(100);
}

void connectToWiFi(const char * ssid, const char * pwd, bool Network){
  

  // delete old config
  WiFi.disconnect(true);
  //register event handler
  WiFi.onEvent(WiFiEvent);

  WiFi.mode( WIFI_AP_STA ); //WIFI_AP_STA     WIFI_AP
  
  WiFi.softAP(AP_SSID); // set AP
  delay(100); // allow time for AP

  //Initiate connection
  if (Network)
     {
     Serial.println("Connecting to WiFi network: " + String(ssid));
     WiFi.begin(ssid, pwd); // 
     }
     else
     {
      Serial.println("Running in Access Point mode only");
     }

  Serial.println("Waiting for WIFI connection...");
}

//wifi event handler
void WiFiEvent(WiFiEvent_t event){
    switch(event) {

      case SYSTEM_EVENT_AP_START:
            //can set ap hostname here
            
            WiFi.softAPConfig( ip, gateway, subnet );
            Serial.println(WiFi.softAPIP());
            Serial.printf("AP [%s] started. IP: [%s]\n", AP_SSID, WiFi.softAPIP().toString().c_str() );
            TCPServer.begin(); // TcpServerPort
            TCPServer.setNoDelay(true);
            Update_Oled_Status(WiFi.softAPIP().toString().c_str());
            break;

      case SYSTEM_EVENT_AP_STOP:
            Serial.println("AP Stopped");
            break;

      case SYSTEM_EVENT_AP_STACONNECTED:
            Serial.println("WiFi AP Client Connected");
            break;
            
        case SYSTEM_EVENT_AP_STADISCONNECTED:
            Serial.println("WiFi AP Client Disconnected\n");
            break;            

        case SYSTEM_EVENT_STA_START:
            //set sta hostname here
            WiFi.setHostname(AP_SSID);
            break;
            
      case SYSTEM_EVENT_STA_GOT_IP:
          //When connected set
          //String MyIP = ""; 
          Serial.print("WiFi connected! IP address: ");
          Serial.println(WiFi.localIP());
          connected = true;
          break;
          
      case SYSTEM_EVENT_STA_DISCONNECTED:
          Serial.println("WiFi lost connection");
          connected = false;
         // udp.stop(); // close UDP Connection
          break;
    }
}`

danbicks, 2018-10-09:
Code enclosed above any ideas or help would be brill.

Big thanks

Dans

danbicks, 2018-10-09:
Handle TCP Section, sorry forgot to include..

`void Handle_TCP()
{

  //if (TCPClient && TCPClient.connected() && TCPClient.available())
  TCPClient = TCPServer.available();
  if (TCPClient.connected() && TCPClient.available())
     {
      while(TCPClient.available()) 
        {
      char c = TCPClient.read();
      TCPInput  += c;
      Serial.println(c, HEX); // only for test
        if (c == '\r') 
           {
            //Serial.println("Carrage Return");
            TCPInput .trim();
            //Process_Telnet(TelnetInput); 
            Serial.print("TCP Data: "); Serial.println(TCPInput);
            //Process_MQTT(TCPInput); // Here we pass message to Process                   
            TCPInput  = ""; // clear string
 
           }

        }
         Serial.println("Test has exited while loop");
     }
}`

lbernstone, 2018-10-09:
The ESPAsyncWebServer can do this. https://github.com/me-no-dev/ESPAsyncWebServer#serve-different-site-files-in-ap-mode. I'm not sure if this would be possible in WebServer, but you could take a look at how it is done, and see if it can be ported.

danbicks, 2018-10-09:
Thanks @lbernstone

I will take a look buddy.

Thanks for your response.

francescolavra, 2018-10-27:
The WiFiServer class doesn't support binding to a specific IP address. Its constructor takes as arguments the port number and (optionally) the maximum number of client connections, so your declaration `WiFiServer TCPServer(TcpServerPort,ip)` is semantically wrong because the second parameter is not an IP address.

chegewara, 2018-10-27:
I dont know if this is an option, but when you get request is there info about client IP? You can add filtering by IP range and respond with error 404 page when is not AP request (just some idea).

francescolavra, 2018-10-27:
Yes, the IP address of a connected client can be retrieved by calling the `remoteIP()` method (which returns an `IPAddress`) on the WiFiClient object that represents a connected client.
But it would be even simpler to query the local IP address for a given client connection, with the `localIP()` method: you compare the returned address with the address of the ESP32 access point interface (192.168.4.1) and you will know whether the client is connected via this interface.

closed 2019-08-15 by stale[bot]