espressif/arduino-esp32 #1950: WiFi.persistent(true) doesn't keep the WiFi.config data.

robertgregor, 2018-10-10:
Hi,
seems like WiFi.persistent(true) works. However, if static IP is used and not the DHCP, then it doesn't keep the config parameters.

My code looks like:
  WiFi.persistent(true);
  WiFi.mode(WIFI_STA);
  if (ip.length() != 0) WiFi.config(getIpAddr(ip), getIpAddr(gw), getIpAddr(mask), getIpAddr(dns));
  if ((ssid.length() != 0) && (pwd.length() == 0)) {
      WiFi.begin(ssid.c_str());
  } else if ((ssid.length() != 0) && (pwd.length() != 0)) {
    if (bssid[0] == 0) {
      WiFi.begin(ssid.c_str(), pwd.c_str());
    } else {
      byte iparr[6];
      for (int i=0; i<6; i++) {
         String part = "0x" + bssid.substring(0,bssid.indexOf(':'));
         bssid.remove(0, bssid.indexOf(':')+1);
         iparr[i] = strtol(part.c_str(),NULL,0);               
      }
      WiFi.begin(ssid.c_str(), pwd.c_str(), chnl.toInt(), iparr);
    }
  }  

Then the ESP 32 will normally join the access point with the defined IP address:

But then, when restart is done I am using:

  WiFi.mode(WIFI_STA);  
  WiFi.begin();
  for (int i=0; i<200; i++) {
    delay(50);
    if (i%20 == 0) Serial.print(".");
    if (WiFi.status() == WL_CONNECTED) break;
  }
  Serial.println();

Then the ESP 32 will join the access point, but use DHCP and will get the IP from the router. So I have to use this code to be able to use static IP: 

  WiFi.mode(WIFI_STA);
  WiFi.config(getIpAddr("192.168.1.50"), getIpAddr("192.168.1.1"), getIpAddr("255.255.255.0"), getIpAddr("192.168.1.1"));
  WiFi.begin();
  for (int i=0; i<200; i++) {
    delay(50);
    if (i%20 == 0) Serial.print(".");
    if (WiFi.status() == WL_CONNECTED) break;
  }
  Serial.println();

francescolavra, 2018-10-26:
Indeed `WiFi.persistent()` is supposed to configure persistence of Wi-Fi-specific settings only, like SSID and password, and not "higher-level" settings such as those dealing with the TCP/IP stack. Adding persistence of such settings wouldn't be trivial, because while for the Wi-Fi settings we can simply call the `esp_wifi_set_storage()` function from the ESP-IDF, there is no analogous function for the IP settings.

robertgregor, 2018-10-30:
Hmmm, OK, I see. Then if it is not the bug, I will close it. Thanks Robert

closed 2018-10-30 by robertgregor