espressif/arduino-esp32 #1951: ESP Async Webserver doesn't serve the files in spiffs larger than 40 kb

robertgregor, 2018-10-10:
Hi, I didn't had problem with the past with this, however, after the latest development release, I have a problem with files, larger than 40 Kb. They are not always served, usually first time, when the access is made.
I can't figure out what is the issue, thus there is no any debug output.
For instance, see the development tools screenshoot:
[image]

As is seen, short files were sent without any problem, but the file jquery-mobile-1.4.5.min.js is not sent within 20 seconds.
When I tried second times, it worked.
[image]

I have the file in SPIFFS filesystem and loaded to the module using the ESP sketch data uploader.

SPIFFS.begin();
server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.html");

This is, how it is in the data dir:
[image]

robertgregor, 2018-10-10:
Hmmm, I've just did more tests, I have i.e. ajax to get some json data. As you see in the network monitor, sometimes the request is very quick, but sometimes it is slow and sometimes it will not even send anything within 20 seconds. Seems, to me some timeing issue in the ASYNC TCP or async webserver, which I am using. See the screenshoot:

[image]

marchingband, 2021-12-15:
I have this issue as well, with this update. I had to write some JS to timeout every request, and retry after a while.
It seems to start off great, then after a lot of requests have been served it slows down, then eventually freezes.