espressif/arduino-esp32 #1953: flash read err, 1000 when ESP_PD_DOMAIN_RTC_FAST_MEM is OFF

Stasikiii, 2018-10-10:
```cpp
#include "esp_deep_sleep.h"
#include <sys/time.h>
#include "time.h"
struct timeval tv;
void printLocalTime(){
  struct tm timeinfo;
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}
void setup(){
  Serial.begin(115200);
  delay(50);
  gettimeofday(&tv,NULL);
  if (tv.tv_sec < 1539178840){
    tv.tv_sec = 1539178840+7200;
    settimeofday(&tv, NULL);
  }
  
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
  esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
  esp_sleep_enable_timer_wakeup(5000000);
  printLocalTime();
  //Serial.println(tv.tv_sec);
  delay(10);
  esp_deep_sleep_start();
}

void loop(){}
```
if   **esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_ON)**;
then it's ok.

BUG?

lbernstone, 2018-10-10:
Not terrific behavior, but documented. https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories

Stasikiii, 2018-10-11:
@lbernstone 
"Note: in revision 0 of the ESP32, RTC fast memory will always be kept enabled in deep sleep, so that the deep sleep stub can run after reset. This can be overridden, if the application doesn’t need clean reset behaviour after deep sleep."
So i have rev 0 of the chip or i don't understand something?

lbernstone, 2018-10-11:
I guess the first sentence applies to rev 0, the second applies to all :)
You need clean reset behavior after the sleep, so you need RTC fast to be active.

Stasikiii, 2018-10-11:
@lbernstone 
I don't understand. :(
When esp going to restart the app just loading from flash is'nt? Why it's need to be loaded in fast mem? Real time clock uses fast mem? but I just need to use the RTC counter..
Ohh.. :(

lbernstone, 2018-10-11:
You may get a better answer from an espressif employee in https://github.com/espressif/esp-idf/issues. I am just another user.
I would guess that the system needs to read something out of the fast memory to boot cleanly.  If you don't care about the clean reboot, perhaps it can be ignored.  How much power does it save to turn off the fast memory domain?

Stasikiii, 2018-10-11:
Thx you @lbernstone 
I try to found it out and post issue later.

Stasikiii, 2018-10-12:
I did ESP32_DEEP_SLEEP_WAKEUP_DELAY 5000
but still 
> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371
:(

lbernstone, 2018-10-12:
That variable is compiled into the libraries.  To change it, you would need to [use arduino as an esp-idf component](https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md)

Stasikiii, 2018-10-12:
I think this is a Hardware issue about flash chip model.

My chip is
HJ1723 (GIGADEVICE)
25Q16CT
AP0T866
Datasheet
I will try Winbond w25q16

Stasikiii, 2018-10-14:
No issue. @igrr recomend enable fast mem... Closed.

closed 2018-10-14 by Stasikiii