espressif/arduino-esp32 #1955: multiple interrupts on same pin esp/arduino

anandpanchal, 2018-10-11:
`In the following code, why does ISR2 never run?

const int in = 19;
const int LED = 9;
int flag;
void setup() {
    pinMode(in, INPUT_PULLUP);
    pinMode(LED, OUTPUT);
    attachInterrupt(digitalPinToInterrupt(in), ISR2, RISING);
    attachInterrupt(digitalPinToInterrupt(in), ISR1, FALLING);

}
void  loop() {
    if (flag) {delay (100); flag = false;}// debounce
}
void ISR1(){
    digitalWrite(LED, LOW);
    // Turn Off the motor, since, Limit Switch was engaged
    flag = true;
}
void ISR2(){ // Limit Switch was disengaged. 
    digitalWrite(LED, HIGH);
    delay(100); // Debounce, so I do not receive spurious FALLING edges from this.
}`

chegewara, 2018-10-11:
I would try to change order and see whats happen:
```
attachInterrupt(digitalPinToInterrupt(in), ISR1, FALLING);
attachInterrupt(digitalPinToInterrupt(in), ISR2, RISING);
```

anandpanchal, 2018-10-11:
If you reordered your code and attached ISR1 before ISR2 then you might see ISR2 run but not ISR1.

chegewara, 2018-10-11:
You see why i did ask that, the conclusion is that second attach call is replacing first one. Why this is happen i dont know, maybe @me-no-dev or someone else can ask this question.

atanisoft, 2018-10-11:
Instead of having two ISRs why not have one that responds to CHANGE and then in the ISR check if it is rising or falling?

apthorpe, 2018-10-17:
Probably to avoid the overhead of a digitalRead() call and the associated logic. But it appears you get a single ISR per pin so using two separate ISRs won't work. It appears that piling all the code into a single ISR that responds to CHANGE is the solution. Reordering the two attachInterrupt() calls shows that the first one is the only one that fires; the second seems to be silently ignored.

Curious if this is a limitation of Arduino compatibility or in the ESP32 itself.

francescolavra, 2018-10-26:
It's a "limitation" of the ESP32 GPIO controller (which in this respect works similarly to the GPIO controllers found in many other MCUs out there): you can configure a GPIO to trigger an interrupt at either rising edges, falling edges, or any edge, but if you configure it to trigger at any edge, when an interrupt is triggered there is no way to tell from the GPIO interrupt status register which type of edge generated the interrupt.

closed 2019-08-15 by stale[bot]