espressif/arduino-esp32 #1960: Can you use ESP8266 Libraries with ESP32?

DanPugly, 2018-10-13:
I was wondering if I could use the ESP8266 Libraries like ESP8266WiFi and ESP8266WebServer with the ESP32?

beegee-tokyo, 2018-10-13:
Will not work.Different CPUs, different hardware, different CPU registers.
Some libraries from the ESP8266 framework are converted to work on ESP32, but if you compare them you can see significant differences how the lib accesses the hardware/OS layers.

Its like you want a Windows program to run on Linux.

Tzelal, 2019-12-07:
I used the <WiFi.h> library for ESP32 but html client didn't respont.The same code works very well with the library <ESP8266WiFi,h> and ESP8266 . Any advice for that ?

diedthreetimes, 2021-05-15:
> Its like you want a Windows program to run on Linux.

Considering ESP32 is a successor to ESP8266 A more accurate comparison may be between iPhone 11 and iPhone 12. 

Many modern programming environments are natively cross-platform. It isn't too difficult to modify the higher layers of the ESP8266 framework to run on either 32 or 8266, and in many cases the platform maintainers have already completed this task.