espressif/arduino-esp32 #1962: VEML6075 I2C sensor not working with ESP32

ladyada, 2018-10-13:
Using an Adafruit Feather ESP32 wired up to an Adafruit VEML6075. A user of our library noticed this here and I verified it with a few different libaries/examples for this sensor.
https://github.com/adafruit/Adafruit_VEML6075/issues/2

It's repeating start and VEML's have had an odd i2c stack so its not terribly surprising
Test code: https://github.com/adafruit/Adafruit_VEML6075
Hardware: https://www.adafruit.com/product/3964
BSP: v1.0.0 from board manager, with Adafruit Feather ESP32 selected (but i really doubt this has to do with the hardware, probably any ESP32 board can be tested)

stickbreaker, 2018-10-14:
Release 1.0.0 does not work when ReSTART operations are used. This error has been fixed in this GitHub dev repo.

When me-no-dev generates a new release the fix will be available through board manager. 

Until then, you can manually apply the fixes by substituting the four i2c files from this repo:
* \cores\esp32\esp32-hal-i2c.h
* \cores\esp32\esp32-hal-i2c.c
* \libraries\Wire\src\Wire.h
* \libraries\Wire\src\Wire.cpp

Chuck.

CheetahGod, 2018-10-14:
thankyou

ladyada, 2018-10-14:
@me-no-dev hey me - is there anything blocking a minor release fix that we can help with? I2c repeated-start sensors are pretty common these days :)

midnightmaker, 2018-10-31:
The above four files are a huge improvement for I2C. There was no communication with an MPU6050 - there is now! It's still very temperamental. I'm using the Doit.com ESP32 Dev Kit V1, trying the standard SDA/SCL pins and alternates, I find the behavior similarly challenged. Sometimes after the sensor has been working for a couple of minutes, the values freeze. At this point the clock and data lines are no longer active on the scope - clock stays high and data is low. At other times, the new firmware recovery works and the sensor communication is restored. Other times, only a hardware reset works. 

Here is an example of a successful recovery:

[D][esp32-hal-i2c.c:1300] i2cProcQueue():  Gross Timeout Dead start=0xc4d5, end=0xc508, =51, max=51 error=1
[E][esp32-hal-i2c.c:314] i2cDumpI2c(): i2c=0x3ffc1030
[I][esp32-hal-i2c.c:315] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:317] i2cDumpI2c(): lock=0x3ffb21a8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:320] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:322] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): event=0x3ffb222c bits=200
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): intr_handle=0x3ffb225c
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): dq=0x3ffb2520
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): queueCount=2
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): queuePos=1
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): errorQueue=2
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:349] i2cDumpInts(): Debug Buffer not Enabled
AcX = -6820 | AcY = -1328 | AcZ = -13344 | Tmp = 26.37 | GyX = -185 | GyY = -32513 | GyZ = 32639
[I][esp32-hal-i2c.c:1091] i2cProcQueue(): Bus busy, reinit
[W][esp32-hal-i2c.c:1363] i2cCheckLineState(): invalid state sda(14)=0, scl(27)=1
[D][esp32-hal-i2c.c:1371] i2cCheckLineState(): Recovered after 1 Cycles

There are other instances where the "Recovered After" Cycle count is 9 cycles, but mostly the count is 1. 
There still seems to be a fundamental problem here. The sensor works flawlessly on my nrf52, Simblee and Arduino UNO boards. It is only on the ESP32 board that the issue appears. Any ideas?

lbernstone, 2018-10-31:
Please don't hijack the thread.  Open a different thread if you have issues with a different device.  This thread is resolved(AFAICT), just waiting for a release version.

stickbreaker, 2018-10-31:
@midnightmaker open a new issue, describe your hardware, when SDA is stuck low, something is driving it.  The reset function in ESP32 I2C will always make the ESP32 stop driving SDA, the MPU6050 is usually the culprit for holding SDA low. With your error being GROSS timeout, this means the MPU6050 didn't answer, try increasing the timeout `Wire.setTimeOut(milliseconds)` the MPU6050 can be programmed to stretch SCL until it finishes a conversion. If the ESP32 times out, the MPU6050 still thinks it is in a read operation.  Increase timeout.

If you want more help open a NEW issue.

Chuck.

midnightmaker, 2018-10-31:
Apologies - I was not looking for help or trying to hijack the thread. I was trying to advise that the behavior of the I2C port was still erratic in a general sense on the ESP32 when compared with other Arduino hardware. However, with the explanation from Chuck above, a change to the timeout does indeed correct the periodic firmware reset behavior. My concern that there was still a general ESP32 I2C problem was misplaced. Thanks for the helpful comment Chuck.

ladyada, 2018-11-07:
hiya @igrr @me-no-dev checking back on this, i noticed no commits in the last 6 weeks - is this repo still going to be developed & supported?

me-no-dev, 2018-11-19:
@ladyada sorry, was preoccupied with other things. Have not really been in one place. I see no reason to throw "is this repo still going to be developed & supported". I am looking for ways to get some help on the repo. Will update IDF and cut a release.

closed 2019-08-15 by stale[bot]