espressif/arduino-esp32 #1964: No library wifimanager ?

moijerem59, 2018-10-14:
Hello, the library wifi manager did not relate to the esp32? If i try the library wifimanager for esp8266 i have compile errors .Thank you
I think it would be interesting to integrate the file library thank you

beegee-tokyo, 2018-10-14:
Try this one [WIFIMANAGER-ESP32](https://github.com/zhouhan0126/WIFIMANAGER-ESP32). I think it has some problems, but it is the version for the ESP32.

moijerem59, 2018-10-14:
I tried this library unfortunately even the example does not compile

lbernstone, 2018-10-14:
The source library at https://github.com/tzapu/WiFiManager works, but you need to use the development branch.

lbernstone, 2019-01-07:
Why?  It is a separate project, and is actively developed there.

closed 2019-08-15 by stale[bot]