espressif/arduino-esp32 #1965: menuconfig flag CONFIG_BLE_SCAN_DUPLICATE

chegewara, 2018-10-14:
@me-no-dev 
Could we switch off this flag in arduino?
`CONFIG_BLE_SCAN_DUPLICATE=y`

I just discovered that turning on this flag is taking almost 50kB from heap for only 10 duplicates and this functionality is duplicated in BLE library with better memory usage because its not required static allocating arrays.

chegewara, 2018-10-15:
@lbernstone 
Never mind, my mistake. I have had to change something else.

But if i may ask something else. esp-idf by default got setup that allows to connect only 3 bluetooth devices at the same time. To make it more generic in arduino, since we cant change it on demand, could we set this value to max allowed?

beegee-tokyo, 2018-10-15:
:+1: 
:+1: 
:+1:

lbernstone, 2018-10-15:
I would think that will require a larger memory allocation.  The base arduino really should be a rapid prototyping tool.  For more significant projects, you should use it as a component or, for best performance, directly use the IDF.

chegewara, 2018-10-15:
@lbernstone I dont know what impact it will have on memory allocation, this is question to espressif devs. 

btw i know its possible to use arduino-esp32 as component, there is nice tutorial how to do it, but i cant see info about other regular libraries that are used in arduino code. Can we use any compatible library that can be used in arduino-ide if we want to use arduino as a component?

closed 2019-08-15 by stale[bot]