espressif/arduino-esp32 #1968: How to tell which version I am using?

carbonadam, 2018-10-15:
I am trying to determine which version Wire I am actually using.  Git can be a bit confusing sometimes so Im not  even sure what command line tool I am meant to run.  Please a little guide so I may check versions.

stickbreaker, 2018-10-15:
@carbonadam this little code segment will display the current `Wire.h` version as an error message during compile.
```c++
#include "Wire.h"

void setup(){
char ch[]=STICKBREAKER;
}

void loop(){}
```
This is what is generated when using the current GitHub Dev Version:
>C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:33:24: error: too many decimal points in number

 >#define STICKBREAKER V1.0.1

 This is what is generated when using Release 1.0.0 (board Manager Version)
>C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\Wire\src/Wire.h:33:22: error: 'V0' was not declared in this scope

 >#define STICKBREAKER V0.2.2
   
Chuck.

carbonadam, 2018-10-15:
Great thanks Chuck it returns 
C:\Users\user\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:33:24: error: too many decimal points in number

 #define STICKBREAKER V0.2.2

---

closed 2019-08-15 by stale[bot]